motioneye-project/motioneyeos #2730: Google Drive upload to same subfolder not possible

ichbaum, 2021-04-28:
Hi,

I have setup the Google Drive Upload on both of the PI's.
I choose the same subfolder for both but in google drive i got 2 seperate folders.

Raspberry Pi 3A+
/surveillance/CAM1/

Raspberry Pi 2
/surveillance/CAM2/

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

Raspberry Cam rev 1.3

[image]

[image]

starbasessd, 2021-04-28:
Are you using the same key for both?

ichbaum, 2021-04-28:
I had different keys.
Now I tried the same key, but then the second cam gets a bad request error.
I requestet a new key for the second cam and now are both in the same folder.

My explanatiopn is, that google did not found the folder, because there was no upload yet.
Because google did not found the folder id created it with an internal ID at first upload. 
If that is at the same time (Testsetup with both cams in the same spot) you got two folders with differend internal IDs at google.
I could see in the link that the hash was different for the folders.
https://drive.google.com/drive/u/0/folders/XXXXXXXXXXXXXXHASHXXXXXXXXXXXXXXXX

closed 2021-04-28 by ichbaum