motioneye-project/motioneyeos #2731: No image displayed from any camera, then the pi hangs

realdeadbeef, 2021-05-02:
### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4 Model B+ (1 Gig Ram)

### Camera

I am using the following type of camera: MMAL - Attached to the Raspi Via the ribbon cable port on the pi (I know it's the right one)

My camera model is: NoIR v2

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am not using any peripherals

### My Problem

Here are the steps that I have followed to get this working on an older version of meyeos:
Open etcher and flash the release for pi4 straight from the release url
Put flashed SD card into Pi 4
Attach power. 
Attach NoIR camera using ribbon cable
Power on Pi
Access the webui from my browser (firefox)
Nothing shows up to start with but I figure this is because I haven't logged in yet and changed any settings
Log in with username: admin
I'm now logged in
Still nothing from the camera
I open settings and change the following settings:
- Camera resolution 1280x720
- Framerate 30

Still nothing
I hit add camera and I get the error: `An Error Occured. Refreshing is reccomended`
So I refresh and the ui no longer loads
The Pi has died

### Log Files

I consider the following log files relevant to this issue:

motioneye.log:
```
2021-04-21 19:00:05: [motioneye]     INFO: hello! this is motionEye server 0.42.1
2021-04-21 19:00:06: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2021-04-21 19:00:06: [motioneye]     INFO: cleanup started
2021-04-21 19:00:06: [motioneye]     INFO: wsswitch started
2021-04-21 19:00:06: [motioneye]     INFO: tasks started
2021-04-21 19:00:06: [motioneye]     INFO: mjpg client garbage collector started
2021-04-21 19:00:06: [motioneye]     INFO: smb mounts started
2021-04-21 19:00:06: [motioneye]     INFO: server started
2021-04-21 19:00:07: [motioneye]     INFO: adding new mmal camera with id 1...
2021-04-21 19:00:07: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2021-04-21 19:00:24: [motioneye]  WARNING: 403 GET /login/?_=1619031622482&_username=_&_signature=f93846380343f1c1769cd24dc4715e2e0a80e6b5 (192.168.0.177) 2.93ms
2021-04-21 19:00:31: [motioneye]    ERROR: authentication failed for user admin
2021-04-21 19:00:31: [motioneye]  WARNING: 403 GET /login/?_=1619031629877&_username=admin&_login=true&_signature=c5d3f8d04846d4c2035376f1e0dc29d4781d5441 (192.168.0.177) 4.28ms
2021-04-21 19:00:49: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2021-04-21 19:00:49: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2021-04-21 19:00:49: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2021-04-21 19:00:50: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2021-04-21 19:00:50: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2021-04-21 19:00:50: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2021-04-21 19:00:50: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
[... 143 lines cut ...]
 * Mounting filesystems: done
 * Mounting overlay filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Configuring CPU frequency: done
 * Starting syslogd: done
 * Loading device-tree overlays: done
 * Starting throttle watcher: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting rngd: done
 * Preparing AP interfaces: ap0
 * Configuring wired network: dhcp
 * Setting current date using SNTP: Sun May 2 07:06:55 UTC 2021
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.0.183/24
 # Default gateway is 192.168.0.1
 # DNS server address is 192.168.0.1
```

starbasessd, 2021-05-02:
If you can SSH in or get to the console (monitor & keyboard) please run the
following commands and report the results:
vcgencmd get_camera
vcgencmd bootloader_version
Also, it is recommended for any Pi4 to use dev20201026, as there are many
issues with 20200606 on Pi4s.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-08:
What is your current status?

starbasessd, 2021-05-28:
No response

closed 2021-05-28 by starbasessd