motioneye-project/motioneyeos #2732: Motion Triggered with No Motion

ShosMeister, 2021-05-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: Pi Zero.

### Camera

I am using the following type of camera: MMAL.

My camera model is: Pi v2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 None

### Log Files

I consider the following log files relevant to this issue:

Let me know if you need a log.

Setup the motion detection and email notification. Get them frequently with 2-4 images attached and nothing is moving. I've tried adjusting settings but think I'm just running in circles.

ShosMeister, 2021-05-05:
I've setup an edited a mask to cover just the smallest area of interest.

I turned on the show changed pixels but they are not shown on the attached images.

starbasessd, 2021-05-05:
If you have Pass Through turned on for movies, turn it off.
If you are not currently using Movies, turn Movies on to see the Show Changed Pixels
If you are using Privacy Mask, you cover the area you don't want to see, ditto with the motion mask.

ShosMeister, 2021-05-13:
Didn't realize the mask was a negative mask. Sorry and thanks! That should do it.

closed 2021-05-13 by ShosMeister