motioneye-project/motioneyeos #2733: I'm experiencing the same problem on my Pi4 8gb As in #2599 as the original poster @xxxTark77xxx .  I've tried several incantations of formatting and installing.  I've used the Pi imager and balenaEtcher, but, no luck.  The boot stops with the same messages as in the first post of this issue.

starbasessd, 2021-05-06:
I'm experiencing the same problem on my Pi4 8gb as the original poster @xxxTark77xxx .  I've tried several incantations of formatting and installing.  I've used the Pi imager and balenaEtcher, but, no luck.  The boot stops with the same messages as in the first post of this issue.

Is there a resolution?

_Originally posted by @rrirower in https://github.com/ccrisan/motioneyeos/issues/2599#issuecomment-833841904_

starbasessd, 2021-05-06:
First, make sure you update the boot loader for the Pi4-8GB, instructions here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
Second, I recommend using Win32DiskImager (available on GitHub)
Third, I would recommend using dev20201026 as 20200606 has numerous issues with Pi4s of all flavors.
If you can, take a snapshot or screenshot of the boot process to try and get the errors.
They may look similar, but may have differences. 
Also, do you have a Linux machine available? It would be handy to check logs.

rrirower, 2021-05-06:
I've uploaded a video that shows what happens at boot.  You can see it [here](https://www.dropbox.com/s/eu9vsw5fk0k7nfk/PXL_20210506_211515094.mp4?dl=0).

I have not tried the steps you indicated above.  Should I go ahead, or, do you want to troubleshoot some more?
I do have an Ubuntu box available.

starbasessd, 2021-05-06:
Yes, first update the bootloader.
Then use dev20201026
Then we'll get the logs.

rrirower, 2021-05-06:
Ok.  Sorry to hear that.  I'll hold off on updating anything until I hear back from you.
Let me know if you need any assistance.

matthew1471, 2021-05-20:
Are you definitely flashing dev20201026 for RPI4 and not another dev20201026?

rrirower, 2021-05-20:
I have not tried it since @starbasessd posted above that it does not work.

starbasessd, 2021-05-20:
Downloaded 
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz
Imaged 16GB MicroCenter Class 4 HC SDCard using Win32DiskImager from
https://sourceforge.net/projects/win32diskimager/
added my wpa_supplicant.conf and ssh.txt files to boot partition, ejected and inserted in Pi4-8GB with 
[root@meye-57b21018 ~]# vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
[root@meye-57b21018 ~]#
Booted normally. 
Added network camera. Changed time zone. Reboot. Started normally.
Can access from either WiFi or ethernet.
I have no clue why it worked, then stopped working, and is now working again on the same Pi4b-8GB with dev20201026 on SDCard. If I get time tonight or tomorrow, I will try with USB boot and dev20201026 and see what's up. 
I won't pull the 'bug' label until after the other test.
Anyone else, please post your results... Be sure to update your bootloader to Sep3 2020 13:11:43

rrirower, 2021-05-22:
I downloaded the dev image and used Win32DiskImager to write to a 64GB XC SD card.  I then wrote the wpa_supplicant file per above.  However, in following the instructions from the [WiFi Preconfiguration page](https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration), I noticed that the sd card does not contain:

_bootloader.bin
loader.bin_

The other listed files are there.  I'm assuming that is not normal.  

The contents of the SD card are posted [here](https://www.dropbox.com/s/x0d8bplkzybntxp/Capture.PNG?dl=0).

starbasessd, 2021-05-22:
RPi4 uses a different boot system, and your listing is correct. Don't forget to add a blank ssh.txt file, too.

rrirower, 2021-05-22:
Success!  I updated the bootloader to April of 2021 using the Raspberry Pi Imager v1.6.1.  I then booted the dev version and it loaded correctly.  I shut down and was able to reboot successfully.

Thanks for your assistance.

starbasessd, 2021-05-22:
Good deal! Glad to help. Don't forget to close when you get a chance.

rrirower, 2021-05-22:
I don't seem to have permission to close the issue.  There's not Close button.

starbasessd, 2021-05-22:
Sorry, my bad. When I shifted it to a new ticket, I became the author. Closing

closed 2021-05-22 by starbasessd