motioneye-project/motioneyeos #2734: Timestamp: two cams right time, one cam wrong time

originalbleak, 2021-05-07:
I confirm that I have read the guide before opening this issue.
I confirm that I have read the FAQ before opening this issue.

### motionEyeOS Version
4.2.2+gitUNKNOWN
motionEyeOS 20200606

### Board Model
RPi Zero W (three)

### Camera
Cam 1: RPI-CAM-V2, vc.ril.camera, MMAL Camera
Cam 2: RPI-CAM-V2 NoIR, http://192.168.X.XX/config/1, Remote motionEye Camera
Cam 3: RPI-CAM-V2 (timestamp problem), http://192.168.X.XX/config/1, Remote motionEye Camera

### Network Connection
All ethernet.
All three cams are attached to the same switch on the same subnet and use DHCP and are the same Hostname.

### Log Files
Don't know which logs are relevant yet.

I've had cams 1 and 2 set up for awhile and recently added cam 3.
I added cams 2 and 3 to cam 1 using the motionEye interface.
The timezone is set to the same zone for all three.
Cam 3 is four hours ahead of the actual time.

I had SNTP selected for all three but tried NTP with server **us.pool.ntp.org** but it didn't help.
I rebooted the system several times and also quit the browser (firefox), flushed the cookies and cache with no change.
I tried a different PC but no change.
Don't know what else to try.
Thank you.

originalbleak, 2021-05-07:
Using these deals...
Plugable USB 2.0 OTG Micro-B to 100Mbps Fast Ethernet Adapter
https://www.amazon.com/gp/product/B00RM3KXAU/

I had a LAN fiasco last month and had to rebuild it so the topmost IPs are not current.

Cam 1
[boot.log.cam1.txt](https://github.com/ccrisan/motioneyeos/files/6444750/boot.log.cam1.txt)
Cam 2
[boot.log.cam2.txt](https://github.com/ccrisan/motioneyeos/files/6444751/boot.log.cam2.txt)
Cam 3
[boot.log.cam3.txt](https://github.com/ccrisan/motioneyeos/files/6444752/boot.log.cam3.txt)

starbasessd, 2021-05-07:
All 3 logs show 3.43/24 for the IP address.
Either you didn't grab the boot.log from the individual Pis or something is corrupt.

originalbleak, 2021-05-07:
Updated OP, my bad.

starbasessd, 2021-05-07:
All 3 logs still show 3.43

originalbleak, 2021-05-08:
I changed cams in the motioneye dropdown menu for each one so what is corrupt?

Ubiquiti ERX router screenshot with IP leases. I've got a TP-Link 5 port unmanaged switch attached to it with a subnet LAN configured. 192.168.3.41 is a fourth cam that I can't get video from yet so I figured that's a totally different issue lol.

[image]

starbasessd, 2021-05-08:
You are grabbing copies of the HUB logs, not the other Pis.
You have to remote into the WebGUI of each pi to grab its logs.

originalbleak, 2021-05-08:
OK, my bad. DUH. I forgot to login to the individual cam to configure it. Assumed that pulling down the menu on one interface configures each cam individually. Maybe someone will have the same problem and this will save them some time. Thank you so much for the support.

starbasessd, 2021-05-08:
Features for Expert Settings and above are only on the local machine, Video
Device and beyond are on ether the local attached camera device or on the
remote device, network camera or remote motioneye camera.
Glad you sorted it out.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-05-08:
Please close if you are working, now.

closed 2021-05-08 by originalbleak