motioneye-project/motioneyeos #2736: images not uploading to Google drive on one of my multiple cameras - all the rest are fine

Hambo1946, 2021-05-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-05-13:
Please answer the questions and include the messages and motioneye logs.

Hambo1946, 2021-05-13:
[messages.log](https://github.com/ccrisan/motioneyeos/files/6473338/messages.log)
I have 4 cameras on Raspberry Pi zeros all working perfectly with the latest version of MotionEyeOS and all are storing images on different windows shares perfectly. However this particular one will not upload anything to Google drive and all the others do. Can anyone offer some suggestions? The camera is in a rather inaccessible place so I would rather not re-image and start from the beginning if I can help it!

Thanks in advance

Graham

starbasessd, 2021-05-13:
The motioneye log is showing it can't find the file to upload.
If you can ssh into the Pi in question, check free space on /data partition (df -h)
Compare the other settings for the gdrive upload to ones that work.

Hambo1946, 2021-05-13:
Hi I looked at the motioneye log and found the entry below. Is this what
you are referring to? I am not very familiar with linux but I am assuming
'/data/media/motioneye_192_168_0_201_public_hambopics_intrays_rpinoir_xbmcuser1/2021-05-13/2021-05-13'
means at there is a link from the raspberry pi to the windows share at
\\192.168.0.201\public\hambopics\intrays\rpinoir. I can see lots of images
using WINSCP on the raspberry pi and assume it is mounting to this
directory. If this is the case, I can confirm that all my other cameras
store their photos on the various subdirectories on the share
\\192.168.0.201\hambopics\intrays and they all work. The share is some
1.16Tb of free space

OSError: [Errno 2] No such file or directory:
'/data/media/motioneye_192_168_0_201_public_hambopics_intrays_rpinoir_xbmcuser1/2021-05-13/2021-05-13'
2021-05-13 16:24:11: [motioneye]    ERROR: failed to upload file
"/data/media/motioneye_192_168_0_201_public_hambopics_intrays_rpinoir_xbmcuser1/2021-05-13/2021-05-13"

starbasessd, 2021-05-13:
I understand that your camera is in a difficult to access location, but my recommendation is to back up the settings, and re-image and restore the settings. 
You didn't post the messages log, and didn't post the df -h for that PI, I cannot continue diagnostics. Maybe someone else can.

Hambo1946, 2021-05-13:
Hi,

Here is a screenshot of df -h. I tried to do df -h > df.txt but it sais it
was a read only filesystem

Hambo1946, 2021-05-13:
[image]

Hambo1946, 2021-05-13:
@starbasessd
There doesn't seem to be a messages.log file but I have renamed the messages file to messages.log. Hope this is what you asked for 
[messages.log](https://github.com/ccrisan/motioneyeos/files/6473646/messages.log)

 there is one called messages which I renamed to messages.log

Hambo1946, 2021-05-13:
[messages.log](https://github.com/ccrisan/motioneyeos/files/6473655/messages.log)

starbasessd, 2021-05-13:
Where do your cameras that work save files?
Settings, File Storage, Storage Device & Root Directory?
How about the Camera with issues?
If you change the Camera with issues to 
Settings, File Storage, Storage Device: Custom Path
and Root Directory: /data/output/Camera1
and leave the File Uploads:  GDrive alone, does it work?

Hambo1946, 2021-05-13:
[messages.log](https://github.com/ccrisan/motioneyeos/files/6473666/messages.log)

Hambo1946, 2021-05-13:
@starbasessd All the cameras store the files as folders onderneath a common location. Here is ascreenshot of the RPI with the issues. All cameras have folders beneath hambopics\intrays

 [
[image]

](url)

Hambo1946, 2021-05-13:
[image]

Here is the screenshot of one of the other cameras with no issues

Hambo1946, 2021-05-13:
I'll have a go at your suggestion of saving the images on the SD card now

Hambo1946, 2021-05-13:
No that drew a blank I am afraid. The pictures were successfully loaded but the Google  part produced no uploads just as before. Screenshots attached
[image]

[image]

starbasessd, 2021-05-13:
Does Test Service work?

starbasessd, 2021-05-13:
And does /RoofCamera/ exist at the root of your GDrive?

starbasessd, 2021-05-13:
Also ssh in, and run 
logrotate -f /etc/logrotate.conf
the repost the logs after trying to upload again.

Hambo1946, 2021-05-13:
Yes the test service works on all cameras. /RoofCamera/ does exist at the root. I have tried to change its name as well on a number of occasions. The test service always works and I can go into GDrive an manually copy images into it but nothing ever is uploaded from MotioneyeOS for this particular camera.

I shall now run logrotate and post you the logs

starbasessd, 2021-05-13:
testing uploads from a PC or MAC doesn't mean it's set up for the Pi.

Hambo1946, 2021-05-13:
No I meant going into Google drive on the PC to see if images were uploaded. I have carried out this course of action for all the otyher pi cameras. It's just this one which has never been able to uploaded iamges to Google.

Here are the new files. I have also rebooted the Motioneyeos

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6474001/motioneye.log)

[messages.log](https://github.com/ccrisan/motioneyeos/files/6474004/messages.log)

Hambo1946, 2021-05-13:
Just to make clear. The Google test service works on all my raspberry pi Motioneye cameras. 
Here is a screenshot showing no files are uploaded to the Google drive folder 
[image]

Hambo1946, 2021-05-13:
@starbasessd 
Yes the test service works on all cameras. /RoofCamera/ does exist at the root. I have tried to change its name as well on a number of occasions. The test service always works and I can go into GDrive an manually copy images into it but nothing ever is uploaded from MotioneyeOS for this particular camera.

I shall now run logrotate and post you the logs

Hambo1946, 2021-05-13:
@starbasessd 
[messages.log](https://github.com/ccrisan/motioneyeos/files/6474357/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6474358/motioneye.log)

Hambo1946, 2021-05-13:
@starbasessd 

Still no luck with the Google upload I am afraid
[image]

Hambo1946, 2021-05-13:
@starbasessd 

Please find the new logs after I tried to upload to Google
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6474385/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/6474386/messages.log)

Hambo1946, 2021-05-13:
@starbasessd 

Looking at the motioneye.log, it seems I have the exact same message when the target is /data/output/Camera1 as I had when the images were stored on my windows share

extract from Motioneyeos.log

2021-05-13 18:48:58: [motioneye]    ERROR: failed to upload file "/data/output/Camera1/2021-05-13/2021-05-13" with service gdrive: [Errno 2] No such file or directory: '/data/output/Camera1/2021-05-13/2021-05-13'
 but that directory does exist and the files are clearly visible in the screenshot below
[image]

starbasessd, 2021-05-13:
I understand that your camera is in a difficult to access location, but my recommendation is to back up the settings, and re-image and restore the settings.

Hambo1946, 2021-05-16:
@starbasessd 

Thank you. I took your advice and did a fresh install which cleared the problem. Thank you for all your help - much appreciated :-)

closed 2021-05-16 by Hambo1946