motioneye-project/motioneyeos #2737: Unable to install on Raspberry PI Zero W - missing documentaion

FLasH3r, 2021-05-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606 / dev20201026

### Board Model

I am using the following board/model: Raspberry PI Zero W v1.1

### Issue at hand

I've downloaded the latest stable / dev image & extracted it to the same path.
Used Etcher to write on the SD card.
Tried with Sandisk, Samsund, Transcend and many others.
Once finished the writing, this is what I see in `diskutil list`

```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *16.1 GB    disk2
   1:             Windows_FAT_16 NO NAME                 31.5 MB    disk2s1
   2:                      Linux                         230.7 MB   disk2s2
```

I see 2 volumes with "no name" in them, listing both of them shows me this:

```
roni@MacBook-Pro /Volumes % sudo ls -la /Volumes/NO\ NAME/   
total 0
d--x--x--x  2 root  wheel   64 May 10 23:27 .
drwxr-xr-x  6 root  wheel  192 May 18 12:04 ..
```

```
roni@MacBook-Pro /Volumes % sudo ls -la /Volumes/NO\ NAME\ 1/
total 13660
drwxrwxrwx@ 1 _unknown  _unknown    16384 May 18 12:04 .
drwxr-xr-x  6 root      wheel         192 May 18 12:04 ..
drwxrwxrwx  1 _unknown  _unknown     2048 May 18 12:04 .Trashes
drwxrwxrwx  1 _unknown  _unknown     2048 May 18 12:04 .fseventsd
-rwxrwxrwx  1 _unknown  _unknown      383 Jun  6  2020 boot-fwupdater.scr
-rwxrwxrwx  1 _unknown  _unknown      331 Jun  6  2020 boot.scr
-rwxrwxrwx  1 _unknown  _unknown    20056 Jun  6  2020 sun8i-h2-plus-orangepi-zero.dtb
-rwxrwxrwx  1 _unknown  _unknown  2772094 Jun  6  2020 uInitrd
-rwxrwxrwx  1 _unknown  _unknown  4175744 Jun  6  2020 zImage
```

putting `wpa_supplicant.conf` in the volume which is not empty does not work. 
I have waited for 15+ minutes in each attempt and that does not work.

Using Catalina 10.15.7.

this is the contents of my `wpa_supplicant.conf` (ssid & passwords are fake here):

```
country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
	ssid="FAKE_SSID"
	psk="STRONG_P@SSW0RD"
	key_mgmt=WPA-PSK
}
```

looked everywhere for a solution.

most articles I found show different contents in the volume - not sure how to solve this (if needed)

https://www.the-diy-life.com/how-to-make-a-raspberry-pi-zero-wifi-security-camera/

starbasessd, 2021-05-18:
You are using the incorrect image if using a PiZeroW.
sun8i-h2-plus-orangepi-zero.dtb is incorrect.
You need:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz
or 
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz
Try those and see if you have issues.
The partition (identifier) you need to put the wpa_supplicant.conf file in is:
1:             Windows_FAT_16 NO NAME                 31.5 MB    disk2s1
If you use Windows, it's the F: Drive

FLasH3r, 2021-05-18:
@starbasessd what a silly mistake 🤦 Thanks!

closed 2021-05-18 by FLasH3r