motioneye-project/motioneyeos #2738: Kamera kann nicht hinzugefügt werden

michihorni, 2021-05-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: IP Kamera Reolink 511W

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-05-18:
https://www.ispyconnect.com/camera/reolink recommends:
rtsp://camera_ip_address/h264Preview_01_sub
When adding Network Camera URL
and putting in username and password into the correct boxes

michihorni, 2021-05-18:
Hallo

ich habe nun ein Update auf motioneyeos-raspberrypi3-dev20201026.img gemacht.

Die Kamera lies sich hinzufügen aber die Einstellungen sind nicht möglich. Siehe Anhang

 

Mit freundlichen Grüßen
Michael Horn

 
 

Gesendet: Dienstag, 18. Mai 2021 um 16:08 Uhr

starbasessd, 2021-05-18:
What index?

starbasessd, 2021-05-28:
What Appendix?

starbasessd, 2022-01-19:
No response in more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd