motioneye-project/motioneyeos #2741: WiFi Passwords With Double Quotes

matthew1471, 2021-05-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model: Raspberry Pi4 4GB.

### Camera

I am using the following type of camera: MMAL.

My camera model is: Raspberry Pi V2.1.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Nothing

### Log Files

I consider the following log files relevant to this issue:

Nothing

### Issue

I have observed that with a WiFi password with " in it, subsequent saves of the configuration results in part of the WiFi password being dropped. For example if the WiFi password is

`IloveFishes"AndCheese`

Then MotionEye will save to wpa_supplicant:

`psk="IloveFishes"

As a workaround I have specified the password as hex using wpa_passphrase (https://www.raspberrypi.org/forums/viewtopic.php?t=240576):

`psk=0xabcbababababa`

starbasessd, 2021-05-21:
There are a number of characters that are technically valid, but cause problems in many applications because they are used as tokens for commands, delimiters, and such. SQL has a number of them, too. It is just something you end up 'working around'.

matthew1471, 2021-05-21:
Agreed, I write programs and understand about needing to "escape" special characters.. but not everyone is me and motioneyeos could be more graceful with special characters.