motioneye-project/motioneyeos #2743: motioneyeos WiFi issues with dev20201026 on Raspberry Pi 4B (4GB)

matthew1471, 2021-05-22:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model: Raspberry Pi 4B (4GB).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Pi Camera 2.1.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * 2 x Ubiquiti UAP-AC HD Access Points

### Log Files

I consider the following log files relevant to this issue:

 * Although I am not getting any crash messages roaming seems to be an issue they're trying to fix : https://github.com/raspberrypi/linux/issues/3849
 
Having an issue where the Raspberry Pi drops off the network.. MotionEyeOS is otherwise still running (the logs report the NAS is down and that it has DNS resolution issues). If I go into the Ubiquiti Network Controller and kick the Pi's connection to the WiFi then motioneyeOS reconnects and all is well again. I suspect there's something dodgy with the WiFi driver. May be linked to roaming / fast roaming which is enabled on the Ubiquiti and seems to be mentioned in the RaspberryPi 4B issues threads.

matthew1471, 2021-05-22:
Going to try "options brcmfmac roamoff=1" in a modprobe file (looks like /data/etc/modprobe.conf should be read) as per https://projectable.me/optimize-my-pi-wi-fi/

starbasessd, 2021-05-22:
Thanks for the report. Until the next dev or full build, your work-around may be to install RaspberryPiOS Lite, and either the motion/motionEye install, or docker and the docker container. The discovery of the 'bug fix/workaround' and updates fixes attempted was after 2020-10-26.

matthew1471, 2021-05-23:
Thanks, looking forward to newer firmware in next release. As a workaround enabling "Connectivity Watch" to the network camera (didn't want to use WAN bandwidth) seems to keep whatever is dying from dying.

matthew1471, 2021-05-25:
Looks like new firmware fixed this : https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=291609

I have frankenstein patched my dev20201026 firmware with newer transplanted firmware and it seems to be working great!

* I deleted fixup.dat and start.elf (firmware files transferred to the VideoCore GPU prior to booting). I instead grabbed fixup4x.dat and start4x.elf from a newer Raspberry Pi OS. I updated config.txt to have "start_x=1" so it will use the alternative filenames (with x'es) instead of the old ones (was easier to see what came from where but presumably if I reused the filenames that would have worked without setting start_x=1)

* I overwrote the bcm2711-rpi-4-b.dtb (RPI 4 device tree) from a newer Raspberry Pi OS.
    Most of the Overlays were identical apart from:
        * sdtweak.dtbo is now built in and configured via the sd_* dtparams in the base DTB. So I deleted that file.
        * spi0-cs.dtbo has been renamed to spi0-2cs.dtbo (so I grabbed that new file too)
        * spi0-hw-cs is "no longer necessary and has been seen to prevent spi0 from working" so just deleted it.
        * 39 other files were modified to reflect the new latest files.

Then booted and got new firmware but same MotionEyeOS Kernel (apparently playing with the kernel without the drivers breaks it)