motioneye-project/motioneyeos #2744: Raspberry Pi4 H.264 OMX Unstable At Higher Resolutions

matthew1471, 2021-05-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model: Raspberry Pi 4B (4GB).

### Camera

I am using the following type of camera: 1 x Network Camera & 1 x MMAL Camera.

My camera model is: Pi Camera 2.1.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * None

### Log Files

I consider the following log files relevant to this issue:

````[63065.926100] ------------[ cut here ]------------
[63065.926120] WARNING: CPU: 1 PID: 5518 at drivers/firmware/raspberrypi.c:63 rpi_firmware_transaction+0xf8/0x134
[63065.926125] Firmware transaction timeout
[63065.926128] Modules linked in: md4 md5 cmac nls_utf8 cifs libarc4 ipv6 nf_defrag_ipv6 brcmfmac brcmutil sha256_generic libsha256 bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) cfg80211 v4l2_mem2mem bcm2835_mmal_vchiq(C) rfkill video>
[63065.926221] CPU: 1 PID: 5518 Comm: kworker/1:2 Tainted: G         C        5.4.51-v7l #1
[63065.926224] Hardware name: BCM2711
[63065.926235] Workqueue: events get_values_poll [raspberrypi_hwmon]
[63065.926239] Backtrace:
[63065.926248] [<c020e0bc>] (dump_backtrace) from [<c020e41c>] (show_stack+0x20/0x24)
[63065.926254]  r7:ffffffff r6:00000000 r5:60070013 r4:c129cba0
[63065.926262] [<c020e3fc>] (show_stack) from [<c0a40a44>] (dump_stack+0xd4/0x118)
[63065.926268] [<c0a40970>] (dump_stack) from [<c022225c>] (__warn+0xe0/0x108)
[63065.926274]  r10:00000000 r9:ef97e540 r8:00000009 r7:0000003f r6:c08a79e4 r5:00000009
[63065.926277]  r4:c0e31920 r3:00000000
[63065.926283] [<c022217c>] (__warn) from [<c022265c>] (warn_slowpath_fmt+0xa8/0xcc)
[63065.926287]  r7:c08a79e4 r6:0000003f r5:c0e31920 r4:c0e31940
[63065.926294] [<c02225b8>] (warn_slowpath_fmt) from [<c08a79e4>] (rpi_firmware_transaction+0xf8/0x134)
[63065.926299]  r8:00001000 r7:e362e700 r6:00000000 r5:ffffff92 r4:ef97e540
[63065.926306] [<c08a78ec>] (rpi_firmware_transaction) from [<c08a7ae0>] (rpi_firmware_property_list+0xc0/0x180)
[63065.926310]  r6:eac0401c r5:00000010 r4:eac04000
[63065.926316] [<c08a7a20>] (rpi_firmware_property_list) from [<c08a7c10>] (rpi_firmware_property+0x70/0x118)
[63065.926321]  r10:e362e70c r9:00030046 r8:ef97e540 r7:efae3ee8 r6:00000004 r5:00000010
[63065.926325]  r4:e362e700
[63065.926332] [<c08a7ba0>] (rpi_firmware_property) from [<bf0aa0c4>] (get_values_poll+0x50/0x160 [raspberrypi_hwmon])
[63065.926337]  r10:00000000 r9:00000040 r8:00000000 r7:eff8a500 r6:eff87300 r5:e33cbc80
[63065.926341]  r4:e37cc2cc r3:00000004
[63065.926349] [<bf0aa074>] (get_values_poll [raspberrypi_hwmon]) from [<c023ebcc>] (process_one_work+0x220/0x534)
[63065.926352]  r4:e37cc2cc
[63065.926358] [<c023e9ac>] (process_one_work) from [<c023ef2c>] (worker_thread+0x4c/0x580)
[63065.926363]  r10:eff87300 r9:c1203d00 r8:eff87318 r7:00000008 r6:eff87300 r5:e33cbc94
[63065.926366]  r4:e33cbc80
[63065.926373] [<c023eee0>] (worker_thread) from [<c0246360>] (kthread+0x144/0x170)
[63065.926378]  r10:e35c5e7c r9:e362ec9c r8:e33cbc80 r7:efae2000 r6:00000000 r5:e362ee40
[63065.926381]  r4:e362ec80
[63065.926387] [<c024621c>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[63065.926391] Exception stack(0xefae3fb0 to 0xefae3ff8)
[... 201 lines cut ...]
[63404.409596]  r4:e37cc2cc r3:00000004
[63404.409605] [<bf0aa074>] (get_values_poll [raspberrypi_hwmon]) from [<c023ebcc>] (process_one_work+0x220/0x534)
[63404.409610]  r4:e37cc2cc
[63404.409618] [<c023e9ac>] (process_one_work) from [<c023ef2c>] (worker_thread+0x4c/0x580)
[63404.409625]  r10:eff87300 r9:c1203d00 r8:eff87318 r7:00000008 r6:eff87300 r5:e33cbc94
[63404.409630]  r4:e33cbc80
[63404.409638] [<c023eee0>] (worker_thread) from [<c0246360>] (kthread+0x144/0x170)
[63404.409645]  r10:e35c5e7c r9:e362ec9c r8:e33cbc80 r7:efae2000 r6:00000000 r5:e362ee40
[63404.409650]  r4:e362ec80
[63404.409657] [<c024621c>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[63404.409663] Exception stack(0xefae3fb0 to 0xefae3ff8)
[63404.409669] 3fa0:                                     00000000 00000000 00000000 00000000
[63404.409676] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[63404.409682] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[63404.409689]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c024621c
[63404.409694]  r4:e362ee40
[63494.661857] CIFS VFS: Close unmatched open
---- booting motionEyeOS dev20201026 ----
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.51-v7l (root@3f131ba76255) (gcc version 9.2.1 20191025 (GNU Toolchain for the A-profile Architecture 9.2-2019.12 (arm-9.10))) #1 SMP Mon Oct 26 01:19:59 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.4
```

starbasessd, 2021-05-23:
Also, if you can SSH in or use the console (keyboard/monitor) run the following commands and report the results:
vcgencmd measure_temp
vcgencmd get_throttled
vcgencmd bootloader_version

matthew1471, 2021-05-23:
It rebooted after a while but :

[root@MotionEye log]# vcgencmd measure_temp
temp=57.0'C
[root@MotionEye log]# vcgencmd get_throttled
throttled=0x0
[root@MotionEye log]# vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685

Am using the official USB-C adapter too.. Was all very unstable when I first got it going.. have been playing with various features to make it more stable (connectivity check to workaround https://github.com/ccrisan/motioneyeos/issues/2743), lower resolution (1024x768) and H.264 (non OMX) to identify what is causing instability.. it looks like there's actually multiple problems.

Just tried raising resolution and using OMX again to get issue almost immediately. Will now try higher resolution but non OMX for debugging. Am hoping is linked to https://github.com/raspberrypi/firmware/issues/1552 which was resolved with newer 4b firmware.

starbasessd, 2021-05-23:
Give it an hour or two and re-run the first 2 commands, please.

matthew1471, 2021-05-23:
With or without OMX?

starbasessd, 2021-05-23:
Either, both.

matthew1471, 2021-05-23:
Just crashed and rebooted again.. 

* Camera 1 (Network Camera 800x600) (OMX)
* Camera 2 (Direct Attached, 1600x1200) (Non OMX)

Will turn off OMX on Cam #1 and see if it it's OMX specific.

[root@MotionEye ~]# vcgencmd measure_temp
temp=57.0'C
[root@MotionEye ~]# vcgencmd get_throttled
throttled=0x0

Will see if this (non OMX) stays stable and run after 1-2hrs as requested :)

starbasessd, 2021-05-23:
Since you are getting random reboots, also post messages, dmesg, and boot logs.

matthew1471, 2021-05-23:
[boot.log](https://github.com/ccrisan/motioneyeos/files/6528373/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/6528374/dmesg.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/6528375/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/6528376/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6528377/motioneye.log)

matthew1471, 2021-05-23:
```
[root@MotionEye ~]# uptime
 17:33:55 up  1:09,  load average: 4.30, 4.31, 4.27
[root@MotionEye ~]# vcgencmd measure_temp
temp=55.0'C
[root@MotionEye ~]# vcgencmd get_throttled
throttled=0x0
```

Performance is obviously not great without hardware encoding.. but stability is perfect!

starbasessd, 2021-05-23:
Have you tried Pass Through on either or both cameras?
I can usually get my cameras up to 1280x720 (PiCams and FosCams)...
Don't know how you feel about doing a build your own motionEyeOS to get the latest kernels and firmware...

matthew1471, 2021-05-23:
Linking in https://github.com/ccrisan/motioneyeos/issues/2043 and https://github.com/ccrisan/motioneyeos/issues/2510

Do you use Raspberry Pi 4? I originally built the SD card for my 8GB Pi4 (hence the dev build).. the Pi Zero has been great.

Was wondering if concurrent OMX was issue but then I think we crashed on 800x600 Camera1.

Could try learning how to build (I'm more Windows dev generally) but Raspberry Pi OS Lite might be quicker for me to test.

Also need to check to see if anyone has reported motioneyeos SMB issue with using the same path.. workaround others used worked of setting share to include directory but not sure multiple mountpoints to the same share just to change the directory is performant.

starbasessd, 2021-05-23:
I do my own RPiOS builds at least every other week. I use my Pi4-8GB or a VM running Debian 32 bit, with a setup called Pi-Gen (available here on github, the 'official build' uses it).
I'm going to try it on a M73 TFF running Windows 10, WSL, and Ubuntu 20.04 and see what kind of results I get (probable fail, due to an issue of building the 32 bit OS on a 64 bit OS...).
I've successfully built motionEyeOS before on the Pi4-8GB.
I closed and locked #1652 because it was 3 years old, and people were posting 'fixed for me's', not requesting help or suggesting other fixes.
I personally point all mine to my NAS (single share point), and include camera name & mp4 or jpg as part of the filename, works for me:
camera1/%Y-%m-%d/mp4/%H-%M-%S
I use whatever the end user requests help on, Pis 0, 0W, 0WH, 1 ,2b, 3b, 3b+, 4b-4GB, 4-8GB. I  don't have any compute modules, though. I have a nice VirtualBox server (Ryzen 7/ 32GB / 10TB) that I can throw almost any OS into. I have a few different cameras (PiCam, but only a PiZero Cable, d-Link DCS-5020-L x3, and FosCam 8800 and RC2) and a nice network setup. My production motionEye hub is another M73/16GB/1TB+NAS, and running Ubuntu Server 20.04. 
I'd like to support all the other Pi-Clones, but finances won't allow.   
camera1/%Y-%m-%d/jpg/%H-%M-%S/%q

matthew1471, 2021-05-24:
`[root@MotionEye log]# uptime
 18:44:38 up  5:36,  load average: 1.86, 2.13, 2.33
[root@MotionEye log]# vcgencmd measure_temp
temp=51.0'C
[root@MotionEye log]# vcgencmd get_throttled
throttled=0x0`

I'm going to try replacing files in the FAT boot-partition and see if the newer release allows OMX.

It crashed about 5 hours while I was trying to view a video file via the GUI.. I will check the logs but I think that was just the watchdog being a bit too aggressive while the GUI grabbed my 600MB file off the NAS over 5GHz. Looked graceful anyway.

Certainly more stable without OMX on the latest dev version.

matthew1471, 2021-05-24:
I have made a frankenstein motionEyeOS boot partition:

- I deleted fixup.dat and start.elf (firmware files transferred to the VideoCore GPU prior to booting).
- I instead grabbed fixup4x.dat and start4x.elf from a newer Raspberry Pi OS.
- I updated config.txt to have "start_x=1" so it will use the alternative filenames (with x'es) instead of the old ones (was easier to see what came from where but presumably if I reused the filenames that would have worked without setting start_x=1)

- I overwrote the bcm2711-rpi-4-b.dtb (RPI 4 device tree) from a newer Raspberry Pi OS.
- Most of the Overlays were identical apart from 
 - sdtweak.dtbo is now built in and configured via the sd_* dtparams in the base DTB. So I deleted that file.
 - spi0-cs.dtbo has been renamed to spi0-2cs.dtbo (so I grabbed that new file too)
 - spi0-hw-cs is "no longer necessary and has been seen to prevent spi0 from working" so just deleted it.
 - 39 other files were modified to reflect the new latest files.

Then booted and got new firmware but same MotionEyeOS Kernel (apparently playing with the kernel without the drivers breaks it)

Giving us :

```
[root@MotionEye ~]# vcgencmd version
Apr 30 2021 13:46:42
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start_x)
```

Will now turn off my WiFi connectivity check (to stop WiFi dying and not reconnecting until the AP kills the connection) and go back to OMX and let's see what is broken/fixed.

matthew1471, 2021-05-24:
System much more responsive and performant generally (even things like booting before I even switched back to OMX).. Am observing motion locking up as per https://github.com/Motion-Project/motion/issues/433 this is likely due to the hardware 1080p limit on the OMX encoder (see https://www.raspberrypi.org/forums/viewtopic.php?p=1762192#p1762192). Most importantly while motion locks up the Raspberry Pi4 is not crashing firmware now!

matthew1471, 2021-05-25:
Have a better test bed:
```

/etc/init.d/S85motioneye stop
ffmpeg -f video4linux2 -video_size 800x600 -framerate 25 -i /dev/video0 -codec:v h264_omx -y -f h264 /dev/null
ffmpeg -f video4linux2 -video_size 1920x1080 -framerate 25 -i /dev/video0 -codec:v h264_omx -y -f h264 /dev/null
```

Testing:

```
[root@MotionEye ~]# ffmpeg -f video4linux2 -video_size 800x600 -framerate 25 -i                                                                                                                                                              /dev/video0 -codec:v h264_omx -y -f h264 /dev/null
ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 9.2.1 (GNU Toolchain for the A-profile Architecture 9.2-2019.12                                                                                                                                                              (arm-9.10)
  configuration: --enable-cross-compile --cross-prefix=/os/output/raspberrypi4/h                                                                                                                                                             ost/bin/arm-none-linux-gnueabihf- --sysroot=/os/output/raspberrypi4/host/arm-bui                                                                                                                                                             ldroot-linux-gnueabihf/sysroot --host-cc='/os/output/raspberrypi4/host/bin/ccach                                                                                                                                                             e /usr/bin/gcc' --arch=arm --target-os=linux --disable-stripping --pkg-config=/o                                                                                                                                                             s/output/raspberrypi4/host/bin/pkg-config --disable-static --enable-shared --pre                                                                                                                                                             fix=/usr --enable-avfilter --disable-version3 --enable-logging --enable-optimiza                                                                                                                                                             tions --disable-extra-warnings --enable-avdevice --enable-avcodec --enable-avfor                                                                                                                                                             mat --enable-network --disable-gray --enable-swscale-alpha --disable-small --ena                                                                                                                                                             ble-dct --enable-fft --enable-mdct --enable-rdft --disable-crystalhd --disable-d                                                                                                                                                             xva2 --enable-runtime-cpudetect --disable-hardcoded-tables --disable-mipsdsp --d                                                                                                                                                             isable-mipsdspr2 --disable-msa --enable-hwaccels --disable-cuda --disable-cuvid                                                                                                                                                              --disable-nvenc --disable-avisynth --disable-frei0r --disable-libopencore-amrnb                                                                                                                                                              --disable-libopencore-amrwb --disable-libdc1394 --disable-libgsm --disable-libil                                                                                                                                                             bc --disable-libvo-amrwbenc --disable-symver --disable-doc --enable-gpl --enable                                                                                                                                                             -nonfree --enable-ffmpeg --disable-ffplay --disable-avresample --disable-ffprobe                                                                                                                                                              --disable-libxcb --disable-postproc --enable-swscale --enable-indevs --enable-a                                                                                                                                                             lsa --enable-outdevs --enable-pthreads --enable-zlib --disable-bzlib --disable-l                                                                                                                                                             ibfdk-aac --disable-libcdio --enable-gnutls --disable-openssl --disable-libdrm -                                                                                                                                                             -disable-libopenh264 --enable-libvorbis --enable-muxer=ogg --enable-encoder=libv                                                                                                                                                             orbis --disable-vaapi --disable-vdpau --enable-mmal --enable-omx --enable-omx-rp            [... cut ...]
[video4linux2,v4l2 @ 0x186a340] ioctl(VIDIOC_STREAMON): Operation not permitted
/dev/video0: Operation not permitted
```

and even crashes the camera:

```
[root@MotionEye ~]# raspivid -l
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates
```

This should be easier to test on RasperryPi OS Lite now! Also we can rule out motion.

matthew1471, 2021-05-25:
Looks like rebooting inbetween testing and 1024x768 runs fine.. suspect it is a resolution problem

matthew1471, 2021-05-25:
Reproduced on Raspberry Pi 3.. 

1024x768 - fine
1920x1080 - error and stacktrace in syslog

```
May 25 14:27:08 test kernel: [  319.894325] bcm2835-v4l2-0: Failed to enable capture port - error -1. Disabling camera port again
May 25 14:27:08 test kernel: [  319.902553] ------------[ cut here ]------------
May 25 14:27:08 test kernel: [  319.902587] WARNING: CPU: 1 PID: 2596 at drivers/media/common/videobuf2/videobuf2-core.c:1548 vb2_start_streaming+0x108/0x178 [videobuf2_common]
May 25 14:27:08 test kernel: [  319.902593] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc brcmfmac brcmutil raspberrypi_hwmon sha256_generic bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l$
May 25 14:27:08 test kernel: [  319.902799] CPU: 1 PID: 2596 Comm: ffmpeg Tainted: G         C        5.10.17-v7+ #1414
May 25 14:27:08 test kernel: [  319.902802] Hardware name: BCM2835
May 25 14:27:08 test kernel: [  319.902806] Backtrace:
May 25 14:27:08 test kernel: [  319.902822] [<809e477c>] (dump_backtrace) from [<809e4b0c>] (show_stack+0x20/0x24)
```

matthew1471, 2021-05-25:
Final thing to play with is h264_omx options:

```
root@test:~/ffmpeg# ./ffmpeg -h encoder=h264_omx
ffmpeg version git-2021-05-24-4c705a2 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --enable-gpl --enable-libx264 --enable-nonfree --enable-mmal --enable-omx --enable-omx-rpi --enable-libfdk-aac --prefix=/usr --extra-ldflags=-latomic
  libavutil      57.  0.100 / 57.  0.100
  libavcodec     59.  1.100 / 59.  1.100
  libavformat    59.  2.101 / 59.  2.101
  libavdevice    59.  0.100 / 59.  0.100
  libavfilter     8.  0.101 /  8.  0.101
  libswscale      6.  0.100 /  6.  0.100
  libswresample   4.  0.100 /  4.  0.100
  libpostproc    56.  0.100 / 56.  0.100
Encoder h264_omx [OpenMAX IL H.264 video encoder]:
    General capabilities: delay
    Threading capabilities: none
    Supported pixel formats: yuv420p
h264_omx AVOptions:
  -omx_libname       <string>     ED.V....... OpenMAX library name
  -omx_libprefix     <string>     ED.V....... OpenMAX library prefix
  -zerocopy          <int>        E..V....... Try to avoid copying input frames if possible (from 0 to 1) (default 1)
  -profile           <int>        E..V....... Set the encoding profile (from -99 to 100) (default -99)
     baseline        66           E..V.......
     main            77           E..V.......
     high            100          E..V.......
```

starbasessd, 2021-05-25:
Just out of curiosity, how much GPU RAM do you have allocated?

matthew1471, 2021-05-25:
Originally 128MB but also tried 256MB

starbasessd, 2021-05-25:
OK, just saw something that said default GPU_MEM settings were way too
small for OMX processing.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matthew1471, 2021-05-25:
Have set to 512MB now.. might be onto something!

Also I read that h264_omx is old, crummy and relies on gpumem but h264_v4l2m2m is the future.

starbasessd, 2021-05-25:
https://www.raspberrypi.org/documentation/configuration/config-txt/memory.md

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matthew1471, 2021-05-25:
With 512MB of RAM the ffmpeg behaves a bit more but motion still gets killed:

```
==> /var/log/motion.log <==
[0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!
```

matthew1471, 2021-05-25:
Final test I suppose is to build a Pi 4 that only accepts network camera feeds and performs OMX encode.. to rule out the camera's partial frames etc being the problem (effectively preventing zero copy too)

wickerman123, 2021-12-05:
Glad I'm not the only one getting issues witth OMX! I'm only just geting started with Pi so can't really delve into details or trying to help resolve issues. 

I was hoping OMX would work as I read that puts encoding on the GPU, currently my main issues are massive framerate drops once we're passed the pre-capture buffer - when motion actually kicks in - which is really annoying. Are there any setup options that allow for a good resolution **and** good framerate? 

Currently running Raspi 4b with NoIR V2.