motioneye-project/motioneyeos #2745: So Lemme get this straight.... the answer was

Riverviewhome, 2021-05-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

Riverviewhome, 2021-05-26:
I don't know how to get motioneye to boot from usb and I don't care if it gets fixed....   is that the answer i read from 8 or 9 months ago from the developer?

starbasessd, 2021-05-26:
It is a work in progress.

Riverviewhome, 2021-05-26:
Thanks.. i just got enlightened to this... I just need a simple viewer for the basement workroom to monitor front door and side porch...  correct assumption that motion eye is the best solution????

starbasessd, 2021-05-26:
Best? Can't say. Watching 2 cameras is very doable. You didn't post any other information from the questions asked in the opening page. You just made a rude comment/question.

closed 2021-05-27 by starbasessd