motioneye-project/motioneyeos #2746: Auto-deletion of pictures

Github197, 2021-05-27:
I setup the motioneye camera to make a picture every x minutes. When the picture is taken it is automatically uploaded to my NAS and would not be needed anymore on the camera. Due to the limited card memory the card storage is filled up quite quickly. I would like to have an option which is automatically deleting the old pictures. It would be enough to have a cronjob running every day to delete all pictures under /data/output/Camera1/ like rm -R /data/output/Camera1/20*, but I am not able to write to the file /etc/crontabs/root.

Is there any other solution how to do it or could there be an option integrated in the web GUI that pictures will be deleted after the successful upload to e.g. FTP?

Thank you, René

starbasessd, 2021-05-27:
If your NAS is capable of SMB/CIFS, you can do the primary creation of the files there by changing: Settings, Files Storage, Storage Device, Network Share, and set the SMB/CIFS share. Then you don't have to worry about storage.
As to running a cron job, SSH or use the console (monitor/keyboard) and follow the instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support

starbasessd, 2021-05-27:
If you have further questions, please answer the questions for the system when you opened the ticket.

Github197, 2021-05-27:
Thank you that helped!

closed 2021-05-27 by Github197