motioneye-project/motioneyeos #2747: Pi 4B can't view camera feed. Web interface OK.

rmfm, 2021-05-30:
Morning all,
Probably me being a complete noob.

On Pi Zero I've had no issue.
New board Pi4B.
Took the direct imagine and flash to SD.
Added new fixup4.dat and start4.elf - https://groups.google.com/g/motioneye/c/-D0LykZI7V8/m/8m0BlvOlCgAJ
Booted in and connected to web interface.
Now I get stuck. I have a ribbon camera attached, but get no image.

I've tried using OS lite and then installing MotionEye, but I run up again little errors.

So as my web interface is working, so clearing MotionEye is active. Why can't I access the ribbon camera?
Any help much appreciated.
Great piece of software.

### motionEyeOS Version
Latest

### Board Model
Pi 4B

### Camera
Board attached
Pi Cam 

### Network Connection

Network cable currently.

### Peripherals
None

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-05-30:
SSH or console (keyboard and monitor) the following commands:
vcgencmd bootloader_version
     needs to be on or after Sep 03 2020. If you need help updating, ask.
vcgencmd get_camera
     Supported should be 1 or more, Detected should be 1. If not post results.
There are more commands needed to run to test, but this is a start.

starbasessd, 2021-05-30:
BTW, for the 8GB model, you need to use dev20201026 here:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz

rmfm, 2021-05-31:
Thanks Camera feed now shows.
New issue is I can't edit any settings... All greyed out "admin" and blank dosen't work.

starbasessd, 2021-05-31:
If you turned on "Fast Network Camera" under Settings, Expert Settings, Turn it off.

rmfm, 2021-05-31:
Sorry it's not that. I haven't even been able to do anything yet :)
[image]

rmfm, 2021-05-31:
However, on my phone no problem. I'll just do it on there.

starbasessd, 2021-05-31:
Which browser? What PC?
Looks like something got corrupted.

matthew1471, 2021-06-04:
Appreciate the topic has moved on now.. but for the benefit of anyone else searching.. the reason your start4.elf and fixup4.dat didn't allow you to view the camera is because those are the versions without the camera drivers, see https://github.com/ccrisan/motioneyeos/issues/2744#issuecomment-847385611 and https://www.raspberrypi.org/documentation/configuration/boot_folder.md

ThePandyMan, 2021-06-20:
> Sorry it's not that. I haven't even been able to do anything yet :)
> Thanks for the advice.
> [image]

I am having this same problem with the Pi 4 on the development version. Settings are greyed out and cannot be adjusted (using chrome on Windows 10). USB camera works on dev, but not on the released version for Pi 4.

starbasessd, 2021-06-20:
@ThePandyMan which Pi4? I have a 4GB and 8GB versions for testing. Let me know and I'll test again.

ThePandyMan, 2021-06-20:
@starbasessd I am using the 8GB board. The steps to encounter this on my end are as follows:

* Boot with camera attached via USB
* Add new camera with the interface
* Apply changes
* Reboot

Once I go through these steps, the settings appear greyed out. Refreshing the page did not seem to have any effect.

starbasessd, 2021-06-20:
Are you using 20200606 and adding the fixup4.dat and start4.elf files? If so, see my notes above and elsewhere. Fixing those 2 files, although allows to boot, breaks USB support. You need to update to dev20201026 which also gives you the ability to boot from USB (change cmdline.txt from /dev/mmcblk0p2 to /dev/sda2)...

starbasessd, 2021-06-20:
Just tested with a generic webcam (TIGA) and works fine
vcgencmd bootloader_version 
shows
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685

ThePandyMan, 2021-06-20:
Initially I used 20200606 and added fixup4.dat and start4.elf files in order to boot. I am now using the dev20201026 version with no changes made on my end. USB is working, but after adding a new camera the web interface breaks.

ThePandyMan, 2021-06-20:
> Are you using 20200606 and adding the fixup4.dat and start4.elf files? If so, see my notes above and elsewhere. Fixing those 2 files, although allows to boot, breaks USB support. You need to update [...]

I will try this.

starbasessd, 2021-06-20:
Don't do an 'update', though, re-image with dev20201026 for Pi4...

ThePandyMan, 2021-06-20:
> BTW, for the 8GB model, you need to use dev20201026 here:
> https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz

Re-flashed boot disk with the quoted disk image file. The USB camera works fine.

starbasessd, 2021-06-20:
Good deal. If you have any other issues, please open a new issue.

closed 2021-06-20 by starbasessd

ThePandyMan, 2021-06-20:
Thank you for your assistance @starbasessd . I found that switching from Google Chrome to Microsoft edge for accessing the web interface fixes the problem with the settings bar being greyed out and non-interactable after making any changes (resolution, password, fps, etc.).

[image]