motioneye-project/motioneyeos #2748: new Raspi 4B 2G doesn't boot

gbi39, 2021-06-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue. 

yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

yes

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: 4B 2GB

### Camera

I am using the following type of camera: V4L2

My camera model is: Logitech C920 HD PRO

### Network Connection

My motionEyeOS unit is connected to the network via: either WLAN or Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

no other peripherals

### Log Files

I consider the following log files relevant to this issue:

no logfiles available, but a screenshot

Problem:

This new Pi hangs straight at the begin of the boot of the OS. 
There are messages on the screen "start4.elf: not compatible" and
"This board requires newer software"
"Get the latest software from: https://www.raspberrypi.org/downloads"

maybe kind of a workaround:

I d'loaded the recent lite version of RPi OS. Stole the start4.elf from there, and voila, the Pi boots.

I had to preconfigure WLAN (otherwise I got a panic and reboot), but then I was able to access the Pi via ssh and http. But my cam was not detected (it works without a problem with a Raspi 3B+ and the corrosponding meye image). So I stay with the 3B+ and meye :)

[image]

celsoazevedo, 2021-06-01:
It seems that the stable build doesn't work on the RPI4 running the latest firmware.

Using dev20201026 worked for me: https://github.com/ccrisan/motioneyeos/releases

starbasessd, 2021-06-01:
You need to update the rpi-eeprom bootloader to at least 09/03/2020, preferably to the 04/23/2021.
You can confirm bootloader version by SSH or console:
vcgencmd bootloader_version
Update/Upgrade instructions here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
Then for the Pi4 the best image to use is dev20201026 here:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz

starbasessd, 2021-06-01:
Also, as an aside, is the USB drive strictly storage, or are you trying to boot to it?
There are instructions posted in the Issues queue for booting Pi4s with motionEyeOS on USB drives... If you'd like to try that, let me know.

gbi39, 2021-06-01:
hi folks

woohoo, you guys are responding very quick.

I'll do as you suggest and report back here asap (can be some days).

Cool project. Thanks a lot so far :-)

starbasessd, 2021-06-01:
> @starbasessd <https://github.com/starbasessd>
>
> meye and all pictures/videos are stored on the sD.
>
> Thanks for your kind offer for assistance :-)

Curious, do you have a USB drive inserted?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

celsoazevedo, 2021-06-01:
I just tested the latest stable on my RPI4 running the latest firmware and it doesn't boot (same error as above). No USB connected, only power, ethernet and HDMI.

No issues with the dev build.

gbi39, 2021-06-02:
@starbasessd 
nope, no usb drive

gbi39, 2021-06-02:
ok, I can give a short interrim stat rep:

- have updated the rpi-eeprom bootloader from 
Feb 16 2021 13:23:36
to the latest
Apr 29 2021 17:11:25
using the recent raspbian-lite and raspi-config

 the meye 20201026 image you folks suggested now boots without a prob. But I havn't been able to attach my cam, cause it's in use.

Will report more asap.

cheers

gbi39, 2021-06-04:
Hi folks,

just an update of my state:

the dev-image mentioned above seems to run good. Cam is detected immidiately. The pi4 is much more reponsive than the pi3b+. I have done no long time runs, so I can't tell anything about stability (stability could be better with the 3b+, had hangs every second day or so)

Nice project. Keep up the good work. Thanks for your kind assistance

starbasessd, 2021-06-09:
Don't forget to close if working, or post an update.

gbi39, 2021-06-10:
works. Thanks for your kind assistance, and the reminder :-)

closed 2021-06-10 by gbi39