motioneye-project/motioneyeos #2749: IP address won't work

cyan-shadow, 2021-06-04:
For the record I am fairly new to any sort of programming

I download Motion Eye Os for my raspberry pi 3 b+
Everything was fine, I was able to get my wpa_supplicant.conf to work after a few tries, then I put in the login and left the password blank which worked. So after all that,  this shows up, [root@meye-14cf87cd ]#.  I don't know what this means but the light from my PI-cam lit up so I know that it works. So I put my IP address from previous lines and all i get is this website from my internet service provider
what do i do?

starbasessd, 2021-06-04:
root@meye-14cf87cd indicates you either used the console or SSH'd in. If you are still at that prompt, you can use the command: 
ip a
to get the ip address, if it is not showing up in the boot up information just above that prompt.

cyan-shadow, 2021-06-04:
I did that and found two possible versions i tried both but all i got was {error page not found}
or that my internet was blocked

starbasessd, 2021-06-04:
What are you using to connect? Win10, Mac, Linux, or phone?
What type of IP address are you getting? 192.168.x.y, 169.254.w.z, or something else?

starbasessd, 2021-06-04:
BTW, the opening questions are there for a reason, it helps us help you so we don't have to play 50 questions.

cyan-shadow, 2021-06-04:
sorry I'm new to this

But it gave me one like this 192. 168.x.yyy/yy, as well as one like this 192.168.x.yyy
I am using a mac, I am connected to the same wifi as I used in the wpa_supplicant.conf 
I am using Opera browser, don't know if that makes a difference, but I also used safari and it didn't connect (it led me to my internet provider's website) 
I also tried using a VPN still no luck
thats everyhting I know and tried

starbasessd, 2021-06-04:
The 192.168.x.y is good. 
What URL are you using in Opera or safari? Should be http://192.168.x.y
From the prompt on the Pi, can you ping the Mac?
Can you open the Terminal on the Mac, and ping the Pi?
ping 192.168.x.y

cyan-shadow, 2021-06-04:
ok
i was just typing in the IP address in the search bar but I'll try adding http://
Also I don't know what you mean by ping

starbasessd, 2021-06-04:
Ping confirms the path and connectivity between 2 computers.
type the command
ping [ip_address_of_Pi]
in the terminal window of the Mac
It will either say Success or Failed
then type the command
ping [ip_address_of_Mac]
will get the same type of reply, Success or Failed
post the results here.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cyan-shadow, 2021-06-04:
ok so I pinged the Pi from my mac and it didn't say success or Failed but there was like a flow of information coming, 64 bytes from [ip address], seq=(its is counting by ones), time=xx.xx ms

cyan-shadow, 2021-06-04:
also when i tried adding https:// in front of the ip address i pinged it said "your internet access is blocked" this is on opera, on safari it said " safari can't connect to server"

starbasessd, 2021-06-04:
It's not https://  only http://

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cyan-shadow, 2021-06-04:
Ok I tried that, I still get the same messages

cyan-shadow, 2021-06-04:
Wow, so apparently it only works on chrome for me. Thank you so much for the help.

closed 2021-06-04 by cyan-shadow