motioneye-project/motioneyeos #2751: MotionEye OS: general settings are not displayed in web interface

floriancartuta, 2021-06-08:
I installed for test the MotionEye OS on Raspberry Pi4B. The problem I encountered is that "general settings" are not shown, only "preferences". I use Picam camera.

How can I enable and access "Advanced settings"?

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
Raspberry PI 4 B

### Camera
Picam
I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290). PiCam

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
UTP cable

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-06-08:
A possibility:
Signed in as 'user' instead of 'admin'

You don't state which version of motionEyeOS. Try re-flashing to dev20201026

floriancartuta, 2021-06-08:
I used this image: motioneyeos-raspberrypi4-20200606.img.xz

starbasessd, 2021-06-08:
Click on the user icon on the upper-left side of the page to switch user to administrator. Use admin with no password when asked for credentials.

starbasessd, 2021-06-08:
I really recommend dev20201026 for Pi4s, especially Pi4-4GB and Pi4-8GB...