motioneye-project/motioneyeos #2752: Having trouble finding the HTML used on the video streaming webpage

CatFisting, 2021-06-09:
On **20200606** using **Pi Zero W**

I'm trying to find where the HTML is that creates the video stream page so that I can modify it to contain some controls for a couple servos on the camera mount.

I know I could put the controls elsewhere, but I'm putting the stream and controls in one place for my mom's camera to keep things simple for her.

Any idea where the file location might be? Thanks for any help.

starbasessd, 2021-06-09:
The html is generated in the python code. Since it is compiled (.pyc files) it's not directly editable. There are tricks available to edit, but that is beyond my level of support. The uncompiled files are available in the source code.

starbasessd, 2021-06-09:
If you do motionEye on another OS, the files can be found at /usr/local/lib/python2.7/dist-packages/motioneye