motioneye-project/motioneyeos #2754: No more video in chrome since update on R91 (fine on Firefox)

kduytsch, 2021-06-09:
I tried the lazy loading image trick, but seems that the flag is no more available on this R91 version

starbasessd, 2021-06-09:
Windows 10:
Google Chrome is up to date
Version 91.0.4472.77 (Official Build) (64-bit)
Works with motionEye 0.41
motion 4.3.2-1
on Ubuntu 20.04 LTS

starbasessd, 2021-06-09:
Same Chrome,
RPi3B+

motionEye Version | 0.42.1
Motion Version | 4.3.1
OS Version | motionEyeOS dev20201026
works fine

starbasessd, 2021-06-09:
If you have any specific set to test, I can do lots of different combinations.

kduytsch, 2021-06-09:
I'll go for a fresh install with motionEyeOS, probably better

kduytsch, 2021-06-10:
ok, motionEyeOS is installed, but impossible to get a static IP. Each time I run the S40Network script, I end up with a DHCP address, any idea ?

starbasessd, 2021-06-10:
Out of curiosity, where are you getting your setup instructions?

kduytsch, 2021-06-10:
I found it easier to do it via GUI indeed. Setup instructions from https://github.com/ccrisan/motioneyeos/wiki/Tweaks#manually-editing-network-configuration

Now, I have GUI up and running, however, I can't import my config from old setup... And can't add my cameras as they say not supported. I was expecting to be able to copy overt the thread<x>.conf file

starbasessd, 2021-06-10:
No, to 'import' you need to do a backup on the old setup, then restore to the new setup, under the WebGUI.
What camera(s) are you trying to set up, and how are you trying to set them up? 
What  Camera Type (Local V4L2, Local MMAL, Network, Remote MotionEye, or Simple MJPEG)
If Network, Remote, or MJPEG, what are you using for the URL?

kduytsch, 2021-06-10:
Import didn't work from the backup, probably because not same version

So I'm trying to re-setup everything from the thread file I have in the backup

All my cameras are IP cam, the URL from the thread are 
netcam_url http://192.168.0.30:88/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2
netcam_url rtsp://192.168.0.31:554/live/ch0
netcam_url rtsp://192.168.0.32:554/live/ch0

starbasessd, 2021-06-10:
Sorry, I don't see camera manufacturer(s) and model(s) on this thread.
The camera setup files are camera-1.conf files in the modern (newer than 20190911)
In motionEyeOS the files are located in /data/etc
It is best to use the WebGUI to add cameras.

kduytsch, 2021-06-10:
well, I'm using cheap chinese camera, no brand there. They have been working since the begining with motion/motionEye

Even on motion website they claim the RSTP format would allow to support any kind of camera

I have now copied the thread.conf into camera.conf and I can see my cameras again. Just seems to be some issue with the "add Camera" command

starbasessd, 2021-06-10:
"Even on motion website they claim the RSTP format would allow to support any kind of camera"
Not exactly. They (motion-project) can support any Valid (follows the standards) RTSP format camera. I have 2 Foscam rtsp cameras and even they don't use the same output URL. One is 
rtsp://ip_address:88/videoMain
and the other is
rtsp://ip_address/video.mp4
when the 'standard' is rtsp://ip_address:554/(path)
Glad to hear you are up. Please close if resolved.

closed 2021-06-10 by kduytsch

starbasessd, 2021-06-10:
Ignore the error, this is just a sample screenshot of how to add your camera
netcam_url rtsp://192.168.0.31:554/live/ch0

[image]

kduytsch, 2021-06-10:
well, I had an issue with libcurl saying rtsp wasn't enabled or so