motioneye-project/motioneyeos #2757: Disable saving pictures completely but still email them? Is it possible?

shandshellin, 2021-06-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: docker container

### Board Model

I am using the following board/model: pi 4

### Camera

I am using the following type of camera:Fast Network Camera

My camera model is: host only with network cameras

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2021-06-11:
Even with your contradictory answers to the questions, the simple answer is  "No".

starbasessd, 2021-06-11:
MotionEyeOS is not an option for motion/motionEye Docker.
Docker is not an option for Fast Network Camera