motioneye-project/motioneyeos #2758: Assign static IP: Network is unreachable

rrirower, 2021-06-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model: Raspberry PI 4.

### Camera

I am using the following type of camera: Smraza Raspberry Pi camera.

My camera model is: SM39.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * USB WiFi adapter

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 

### What is the proper way to configure for static IP?

I had my system working properly and decided to add a notification hook.  That worked correctly, but, as part of the process, I decided to change the IP for my PI to a static IP.

I was already logged onto the camera system and opened my router software.  I changed added the PI with a static IP.  I then changed the MotionEyeOS to reflect the static IP.  I rebooted the device.

Now, I am stuck in a boot loop with the message "Network is unreachable".  The boot does show the correct IP address.

Do I have to reflash to resolve this?  If so, what is the proper way to set up a static IP?

starbasessd, 2021-06-12:
There are several ways to set a Static IP address in motionEyeOS:
My preferred: 
`     Set your DHCP service to issue a static ip address to your Pi's MAC address.`
MotionEyeOS Option 1:
`     Open the WebGUI, Settings, Network, Change IP Configuration to Manual, and enter your information.`
MotionEyeOS Option 2:
`     Create a file 'static_ip.conf' in /boot when you first flash the SDCard. Include the following
in the file:`
`STATIC_IP=192.168.X.Y/24`     `### Change X and Y  to numbers appropriate to your network`
`STATIC_GW=192.168.X.Z`       `### Change X to your subnet, and Z to your Router IP Address`
`STATIC_DNS=192.168.X.Z`     `### Change as appropriate for your router `
     More instructions here: 
     https://github.com/ccrisan/thingos/wiki/static_ip.conf`

rrirower, 2021-06-12:
How do I open the webui when the system is stuck in a loop?

Again, is my only option to reflash?

starbasessd, 2021-06-12:
BTW, each of the above methods are mutually exclusive. Only use 1.
As to recovery, yes, re-image is best.

rrirower, 2021-06-15:
For anyone else that runs into this issue of trying to set a static IP after running the system for a while, here's what I did:

1. Obtain the MAC address of my Raspberry Pi.
2. Update my Router settings to reflect a static IP for the Raspberry Pi.
3. Reflash the SD card for the Pi with MotionEyeOS and copy the necessary files to it.
4. Reboot the Raspberry Pi.
5. The Pi was assigned the static IP I had selected.

I will check in about a week to verify that this still works.

closed 2021-06-15 by rrirower