motioneye-project/motioneyeos #2759: Pi Zero W + Fast Network Camera + RTSP = no color, greyscale

icedan, 2021-06-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

motionEyeOS 20200606

### Board Model

Raspberry Pi Zero W

### Camera

I am using the following type of camera: Fast Network Camera and Simple MJPEG Camera

My camera model is: Pi ZeroCam OV5647 5MPx

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I have a problem with my MotionEyeOS and my camera in fast network mode. If there is MJPEG the camera has normal colors, but if it goes to RTSP it is grayscale. Am I doing something wrong?

jasaw, 2021-06-21:
@icedan Please check your camera configuration after switching to RTSP as some settings may change. Double check your white balance to make sure it's not set to grey scale.

icedan, 2021-07-09:
Ok, problem was with to low bitrate.

closed 2021-07-09 by icedan