motioneye-project/motioneyeos #2761: Network with no internet using evertz Master clock

BrandonHortman, 2021-06-22:
Preliminary Docs

I confirm that I have read the CONTRIBUTING guide before opening this issue.
I confirm that I have read the FAQ before opening this issue.

motionEyeOS Version

motionEyeOS 20200606

Board Model

Raspberry Pi Zero W

Camera
My camera model is: Pi v2

Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I manage a network at an Olympic stadium. The network doesn't have internet connectivity,  but the server gets NTP from an evertz master clock in our production studio. 

After working with it for a week and reading the log files,  I believe I'm not getting time from the server,  since it's not 1970 of course lol

After connecting to the wifi, getting an ip, it tries to initialize the eth0 (not connected)  then panics.

Is there a workaround for this?

starbasessd, 2021-06-22:
Since you can't access Settings, Expert Settings, you can create a file: /boot/date.conf (similar to the /boot/wpa_supplicant.conf file with your pre-boot wifi settings)
You would include the lines:
-----
DATE_TIMEOUT="10"  << in seconds, extend if needed
DATE_METHOD="ntp"   << see bellow
DATE_NTP_SERVER="www.xxx.yyy.zzz:pppp"   << ip address and port of ntp source
-----
I do not know if you are using internal DNS resolution so I suggest the ip address. If you are using DNS, you could replace the IP addresses with the name, like time.stadium.local (or whatever)
Full description here:
https://github.com/ccrisan/thingos/wiki/date.conf

BrandonHortman, 2021-06-23:
Thank you for your reply. 

Maybe I am wrong about my diagnosis, because it still seems to panic. 

 Looking in my message log I see this: 

user.notice network: no default route, calling panic action

[image]

Boot log: 

[image]

starbasessd, 2021-06-23:
It would appear that your DHCP server is not giving a default gateway, or, if you are setting a static, you did not set the default gateway.

starbasessd, 2021-06-23:
For diagnostics, you also might change the watch status: 
create a file:
/boot/watch.conf
put the following lines in it:
-----
LINK_WATCH="false"
IP_WATCH="false"
-----
This should have the effect of stopping the panics, but at the cost of not knowing if networking is failing. If you have console access (monitor & keyboard) this would at least bring the system up to check other things.

starbasessd, 2021-07-19:
What is your status?

BrandonHortman, 2021-07-19:
My apologies. I just had a baby! 😃
You were also dead on, I did not have a gateway. Everything works fine. 

Thank you

starbasessd, 2021-07-20:
Good deal, congratulations, and the next 18 years will be very interesting...  :)

closed 2021-07-20 by starbasessd