motioneye-project/motioneyeos #2762: fails to boot on raspberry pi 400

marc1uk, 2021-06-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026

### Board Model

I am using the following board/model: raspberry pi 400

### Camera

I am using the following type of camera: V4L2 (USB)

My camera model is: Easing B08DR3QMBR

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

I consider the following log files relevant to this issue: None (it doesn't make any)

### Description

Loaded a 32GB SD card with the dev version of MotionEyeOS, which previous issues report should work OK for rpi 4. I get the rainbow, followed by a series of terminal printouts all of which report "done" - no error messages. Then the pi just turns off. This happens very quickly - I had to record the screen with my phone in slow-mo video capture to even check for error messages.

The board and SD card work fine with manjaro arm.

starbasessd, 2021-06-25:
Can you post a screenshot of video right before it shuts down? There are numerous differences between the Pi4s and Pi400.
Can you load RaspberryPiOS, and post the results for the command
vcgencmd bootloader_version
please? Have you done the 
sudo rpi-eeprom-update -a
command? Or done an RPi Imager rpi-eeprom-update SDCard creator image?

marc1uk, 2021-06-29:
Sorry that this might be a slow issue, I'm afraid I'm very limited on time right now. I've currently set my pi400 up on manjaro, just to check everything was working before I opened the issue (i saw on the rpi forums that 32GB SD cards could have issues, but it's working just fine). I'd rather not nuke that, especially as i plan on picking up some other cards, so i'll post back when i've had a chance to grab a new SD card and do the tests you suggest.

marc1uk, 2021-07-23:
Right, apologies again for the slow response. I grabbed a fresh copy of RaspberryPiOS and ran the suggested command:
```
$ vcgencmd bootloader_version
Jul 16 2020 16:15:46
version 45291ce619884192a6622bef8948fb515c2b456 (release)
timestamp 1594912546
update-time 0
capabilities 0x00000000
```
then as suggested
```
$ sudo rpi-eeprom-update -a
bootloader update available
current 16 Jul 2020
latest 29 Apr 2021
VL805_FW using bootloader EEPROM
VL805 up to date
```
seems like this did an update,so i rebooted and `vcgencmd bootloader_version` indeed indicated the change, and `sudo rpi-eeprom-update -a` this time reported up to date. 
Back to motioneyeOS, unfortunately this seems to have taken a step backwards - the green light blinks once or twice in quick succession (strictly twice but it's a little hard to tell) followed by one long blink, with nothing on the HDMI out. Then it just turns off. The [rpi website](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md) doesn't even seem to suggest a single long flash is associated with an error. Going by the old threads I replaced `fixup.dat`and `start4.elf` with those from the RaspberryPiOS just out of curiosity (shouldn't be needed and won't work, but maybe it'll provide debugging info), and it did get a little further -- i got to the rainbow......... and still had the rainbow several minutes later. 
Manjaro still working fine after the EEPROM update.

starbasessd, 2021-07-23:
Unfortunately is possible that we'll need the next update / release of motionEyeOS for the Pi400 to be supported, as it came out about 10 days after the last dev release.

marc1uk, 2021-07-24:
OK, I'll try to keep an eye on this repo for the next update and try again when that gets released. Thanks for the support and continued development. :+1:
I suppose I'll leave this issue open for visibility to others with a Pi 400, but feel free to close it if desired.