motioneye-project/motioneyeos #2763: First time trying motioneyeos with raspberry 4, and wont boot.

Ruissilva, 2021-06-25:
I have a raspbery pi 4 2gb, with a 4gb sd card. I followed the steps for a windows install but im having this issue:

https://github.com/ccrisan/motioneyeos/issues/2419

I read the comments but its not clear to me what i have to do. It says here i need the files wpa_supplicant.conf and ssh files, fixup4.dat & start4.elf. I download a zip file, replaced the  fixup4.dat & start4.elf with the existing ones and its even worse now, so i installed the image again. same error, so now i guess i need the  files wpa_supplicant.conf and ssh files, so where can i find these files?

really poor first impression btw... and judging by the number of issues... really hope its only a bad first impression, fingers crossed

starbasessd, 2021-06-25:
In more recent issues, there is better documentation as to what needs to be done.
First, you need to bring your Pi4's eeprom up to current. The RPi Foundation originally had many issues with booting, and was missing many features that were available on the Pi3's. Instructions for updating the Pi are here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
Once the RPi-eeprom has been updated to April 29 2021, confirm with the command
vcgencmd bootloader_version
You then use dev20201026 found here:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz
No need to replace the fixup4.dat or start4.elf files.
As to wpa_supplicant.conf, and ssh.txt, these are files that Raspberry Pi OS also needs to enable wireless and ssh operation, and are installed the same way, and created the same way. There are instructions here:
https://www.raspberrypi.org/documentation/configuration/wireless/headless.md
For SSH:
https://www.raspberrypi.org/documentation/remote-access/ssh/README.md 
Scroll down to section 3...
Sorry to hear you were disappointed. Hopefully this helps. The number of issues is related to the number of different different type of boards supported, running as absolute minimal systems, etc. There are 12 different boards, 6 different manufacturers, with 12 different processors, single core, multi core, RAM limits, etc. with limited resources for support. I try. I have most of the RPi boards, I have a VirtualBox server to create the setups for PC users. I do this for free. Mr Crisan is doing the best he can.

Ruissilva, 2021-06-25:
And it worked :D looks great! thz for the help, great support.

closed 2021-06-25 by starbasessd

lrdqs, 2021-08-28:
Good evening. I have the same issue and can't get it to work. Using a Pi 4 (4 GB) and a 32 GB SD card. Updated bootloader using Pi Imager to April 29 2021, copied the same wpa_supplicant.conf and ssh files that work with PiOS to the 30 MB partition (containing config.txt, kernel.img, start.elf, ...) However, the device does not appear in the network overview. Any help appreciated.

starbasessd, 2021-08-28:
How are you searching for it? It does not show in Network Neighborhood.
Win 10 use IPScanner or AngryIP scanner... 
If you can hook up a monitor, it shows the IP address on the boot screen.

lrdqs, 2021-08-28:
I checked my router first. Set up MotionEye on a Pi Zero before and it automatically showed up there.

I'm on Linux. Using Nmap to scan the network doesn't show the Pi 4, either.

starbasessd, 2021-08-28:
I would then need to hook a monitor up to see if it is going into a panic boot loop. 
If you have another place in the location where you can hook it up to ethernet, try that. 
You are using image:
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi4-dev20201026.img.xz

lrdqs, 2021-08-28:
I would need a Micro HDMI adapter or HDMI --> Micro HDMI cable for that, right?
Connecting directly to the router via Ethernet also doesn't show the device in my network.
Yes, I'm using motioneyeos-raspberrypi4-dev20201026. Extracted the file so it's .img and used Pi Imager to write it on a SanDisk Extreme 32 GB SD card. Then mounted the boot partition, pasted the two files and unmounted.

starbasessd, 2021-08-28:
Interesting.  
Cable or adapter = Yes. 
Sounds like a no network panic loop but the Ethernet should work. 
Can you try a different copy of the wpa_supplicant.conf file?
\# - - - - -
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID-1"
    psk="p@$$W0RD"
	key_mgmt=WPA-PSK
}
\# - - - - 
Change country, ssid and psk as appropriate, in quotes as shown.
If possible, if using Linux, use nano for the editor to create it.
If Win10, I recommend NotePad++
You say the Pi4 boots fine to RaspiOS and is visible on the network?

lrdqs, 2021-08-28:
Tried as you said, created wpa_supplicant.conf and ssh using touch/nano on a fresh installation but it won't work. Also tried just installing the image and connecting via Ethernet - no success.
Yes, PiOS worked like a charm and was visible on the network.
This is odd.

starbasessd, 2021-08-28:
If you remember the IP address it got running RaspiOS, it should pull the same one. Try pinging it when the motionEyeOS is installed.

lrdqs, 2021-08-28:
I do. Using nmap IP-address leads to "Host seems down." nmap -Pn IP-address --> "Host is up (0.037s latency)."
Can't get to the MotionEye UI via Browser, though.

starbasessd, 2021-08-28:
-Pn: Treat all hosts as online -- skip host discovery

starbasessd, 2021-08-28:
I don't know. If you can connect to the Pi4 when it's running RPiOS, it should be working with motionEyeOS dev20201026 and the same wpa_supplicant.conf or ethernet connection. 
With the RPiOS running, connect via ssh, can you run the following commands, and report the results?
ip a
results?
sudo rpi-eeprom-update
results?
sudo rpi-eeprom-update -a
results?
reboot
sudo rpi-eeprom-update
results?

lrdqs, 2021-08-28:
Oh, ok... my bad.
Strange that it doesn't work. Tried different SD cards, Pi Imager, Etcher and Win 32 Disk Imager, Wifi-Setup, Ethernet and nothing works.
Will try to get a Micro HDMI cable/adapter on Monday I guess.

lrdqs, 2021-08-28:
**ip a**

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether e4:5f:01:52:e7:98 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:52:e7:99 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.25/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863928sec preferred_lft 755928sec
    inet6 2a02:8108:96bf:d688:b81c:d143:35d8:ed2/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 7123sec preferred_lft 3523sec
    inet6 fe80::96a0:fc68:811e:877c/64 scope link 
       valid_lft forever preferred_lft forever

**sudo rpi-eeprom-update**

BOOTLOADER: up to date
   CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
    LATEST: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

**sudo rpi-eeprom-update -a**

BOOTLOADER: up to date
   CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
    LATEST: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

**reboot**

Failed to set wall message, ignoring: Interactive authentication required.
Failed to reboot system via logind: Interactive authentication required.
Failed to open initctl fifo: Permission denied
Failed to talk to init daemon.

**sudo rpi-eeprom-update**

BOOTLOADER: up to date
   CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
    LATEST: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

starbasessd, 2021-08-28:
And you used the same ssh.txt and wpa_supplicant.conf file?
Very strange. 
You could try adding a static IP address file to the motionEyeOS /boot folder:
https://github.com/ccrisan/thingos/wiki/static_ip.conf
Filename: static_ip.conf
\# - - - -
STATIC_IP="192.168.178.25/24"
STATIC_GW="192.168.178.1"
STATIC_DNS="8.8.8.8"
\# - - - -
and try that...

lrdqs, 2021-08-28:
Yes, I used the exact same files.
Static IP also doesn't do the trick. Still can't find the device in my network.

starbasessd, 2021-08-28:
Let me know when you can see it booting...

lrdqs, 2021-08-28:
Yes, will try to get an adapter or cable asap.
Thanks a lot for your help so far!