motioneye-project/motioneyeos #2764: Can't seem to connect to web interface

macleancampbell, 2021-06-25:
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

I am running motionEyeOS version: 20200606

I am using the following board/model: Raspberyy Pi Zero W

I am using the following type of camera: Raspberry PI CSI camera

My camera model is: Raspberry PI CSI camera

My motionEyeOS unit is connected to the network via: wifi

I am unable to connect to the web interface in order to access the MotionEye OS on my Raspberry Pi Zero W.

I can see the connection on my router, the pi zero w has an IP address, but I cannot ping it from my Windows 10 PC, and I cannot access it through a web browser using the IP address it is connected with (192.168.1.10).
"The server at 192.168.1.10 is taking too long to respond."
"Pinging 192.168.1.10 with 32 bytes of data:
Reply from 192.168.1.5: Destination host unreachable.
Reply from 192.168.1.5: Destination host unreachable.
Reply from 192.168.1.5: Destination host unreachable.
Reply from 192.168.1.5: Destination host unreachable.

Ping statistics for 192.168.1.10:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),"

I am also unable to assign the MotionEye OS Pi Zero W a static IP address using static_ip.conf. At first, the static IP works, connecting to my router with the IP I specified, however, after a few minutes, it reverts back to the most recent automatically assigned IP address.

So far, I can't get the system to actually work in any way that I can use it.

starbasessd, 2021-06-25:
What is the IP address of the Windows 10 PC?
On the PiZero, how are you connected? Console (keyboard & monitor?
Can you assign a static IP address via the Router and the PiZero MAC address?

starbasessd, 2021-06-25:
One additional question:
Wht IP address does it revert to?

macleancampbell, 2021-06-25:
> 
> 
> What is the IP address of the Windows 10 PC?

192.168.1.5

> On the PiZero, how are you connected? Console (keyboard & monitor?

I don't have any hard connections to the Pi Zero W. Unfortunately, I lack the adapters necessary. I ordered the wrong HDMI adaptor, and would be using a bluetooth keyboard, but only if I had a monitor as well.

> Can you assign a static IP address via the Router and the PiZero MAC address?

I just did that, and again, it defaulted to the auto assigned one.
> 
> 
> One additional question:
> Wht IP address does it revert to?

192.168.1.10     I've been assigning it to 192.168.1.40 unsuccessfully.

starbasessd, 2021-06-25:
And you can't ping it at either address?

starbasessd, 2021-06-25:
What are you using to create the static_ip.conf address? Don't use notepad or wordpad. There is a nice, free program called NotePad++. Also make sure you can see file extensions, and that the program you use doesn't default it to static_ip.conf.txt

macleancampbell, 2021-06-25:
> 
> 
> And you can't ping it at either address?

I cannot. I get the same result whether it is 192.168.1.10, or 192.168.1.40

> 
> 
> What are you using to create the static_ip.conf address? Don't use notepad or wordpad. There is a nice, free program called NotePad++. Also make sure you can see file extensions, and that the program you use doesn't default it to static_ip.conf.txt

That is the program I used, and I changed the file extension accordingly.

starbasessd, 2021-06-25:
And the contents of the file when placed in /boot/static_ip.conf (along with wpa_supplicant.conf)
-----
STATIC_IP="192.168.1.10/24"
STATIC_GW="192.168.1.1"
STATIC_DNS="192.168.1.1"
-----

macleancampbell, 2021-06-25:
> 
> ## And the contents of the file when placed in /boot/static_ip.conf (along with wpa_supplicant.conf)
> ## STATIC_IP="192.168.1.10/24"
> STATIC_GW="192.168.1.1"
> STATIC_DNS="192.168.1.1"

yes, except for the "/24" is that essential? I didn't know.

starbasessd, 2021-06-25:
Yes, it's the equivalent of NetMask (for a 192.168.1.x network would be 255.255.255.0)

starbasessd, 2021-06-25:
You can skip the STATIC_DNS but for routing on your network, you DO need the STATIC_GW also. It's a point on the network that ALL the devices are able to connect to... You would only skip it if you only had the Pi on the network and didn't need the internet connectivity.

macleancampbell, 2021-06-26:
Ok, so I updated the static_ip.config with your suggestion, for testing's sake using 192.168.1.10 in both the pi's static ip and in the router's settings for the pi's mac address..... And I have access to the web interface now! Thank you for your help. I had been following your advice in: https://github.com/ccrisan/motioneyeos/issues/2558 and it had gotten me to this point, and now you've got me online completely. Thank you!

closed 2021-06-26 by macleancampbell

starbasessd, 2021-06-26:
Sorry I didn't have the /24 in that one, I corrected it, too...