motioneye-project/motioneyeos #2766: help and opinion about start recording when one variable is 1.

Ruissilva, 2021-06-29:
Hello,
I have a security system manadged by an arduino. I have a boolean signal in the arduino code that indicates if there is a robbery or not. I Want to send that variable to the raspberry and start camera recording when this variable is 1. What is the best way to do this?

starbasessd, 2021-06-29:
There are many ways to do so, but no 'best' way. 
If your arduino has access to the intranet (ethernet or wifi), you could send a command via the web interface:
enable connections to motion web interface:
nano /data/etc/motioneye/motion.conf
change line webcontrol_localhost from on to off
send the following: wget ip_address_of_motioneye:7999/1/action/eventstart
There are several commands you can use, like wget, curl, lwp-request, etc...
There are ways to create python scripts to do similar based on GPIO pins for the Pis

starbasessd, 2021-06-29:
Info on Web Control:
https://motion-project.github.io/motion_config.html#OptTopic_Webcontrol

Ruissilva, 2021-06-29:
Ok, i'm thinking on following this example for arduino and raspberry comunication, using a usb cable. 
https://www.youtube.com/watch?v=nh5geiIDqjA

I'll create a script to aquire the variable i want, but i don't know how i can use this variable with motioneyeos. Sorry, i'm only starting to use motioneyeos now. motioneyeos don't have the option to control recording by a variable, so maybe i can do this by a script..

Ruissilva, 2021-06-29:
Oh juat read the line: There are ways to create python scripts to do similar based on GPIO pins for the Pis.

starbasessd, 2021-06-29:
Here's a bunch of scripts to control lights, and such, that can be modded to use your variable:
https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons
However, using a USB connection may be problematic, as the motionEyeOS is locked down quite a bit. You may want to consider using motion/motionEye from here:
https://github.com/ccrisan/motioneye/wiki
to get full use of an OS and USB.

Ruissilva, 2021-06-29:
Ok, thanks for the help. I'll advance a bit in my project and then give news. :)