motioneye-project/motioneyeos #2768: include gpiozero

kingtat, 2021-07-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

I'd like to attempt some gpio programming, with led lighting and a pir. gpiozero seems to make this much easier for a beginner.
I wonder if gpiozero library might be included in the motioneyeos build. Alternatively, I'd be grateful for pointers on building motioneyeos with gpiozero package. I've tried installing motioneye on standard raspbian lite. However performance is not as good as motioneyeos, and also the fast webcam option appears to become unavailable when I go this route.

starbasessd, 2021-07-04:
Please confirm that the requested library is missing in dev20201026.
Do you need python2 gpiozero or python3 gpiozero?
Depending on depends, have you tried (for python3) copying the gpiozero folder from a working Pi to your PiZero? 30,000ft overview:
Enable debug (Settings, Expert Settings) and reboot (or remount / as RW)
scp -r pi@working_ip_address:/usr/lib/python3/dist-packages/gpiozero /usr/lib/python3/dist-packages
Turn off Debug (or reboot if you did the mount RW option)
try your script(s)

kingtat, 2021-07-05:
Thank you for your help  starbasessd.

dev20201026 does not include gpiozero. I did as you suggest and copied over the gpiozero package contents (from raspbian lite) directly to the sdcard after writing dev20201026 image. I placed these in `python2.7/site-packages` since dist-packages doesn't appear to be in the python path. There are some additional dependencies such as **colorzero** & **signal**.
I'd prefer to go the build root. i.e. Add these new dependendencies and rebuild motioneyeos image via docker. But I do not know how to add the additional python packages to the build configuration. How do I do that?

starbasessd, 2021-07-05:
Building the system is beyond my level of support, other than to point you at the instructions for doing so, here:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
You would add the needed python libraries during step 3.

kingtat, 2021-07-05:
Thanks again.
Copying just the package contents over, I do have gpiozero working now. An instance of MotionSensor works at any rate.
I'd still like to try building the OS image with additonal python packages, so will give that a go some time soon.
Cheers.

starbasessd, 2022-01-19:
No response in more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd