motioneye-project/motioneyeos #2770: Admin PW issue

NickRudd90, 2021-07-08:
Sorry for the lack of details here on my phone
Running the latest version of motionEyeOS for raspberry pi 3 b+
Freshly installed.
When plugging directly into the pi I can login to admin with no password (Black background white text). When on the GUI I put in admin as the user no password and I get an incorrec password. I just can't seem to get in? Any help would be much appreciated.

starbasessd, 2021-07-08:
The command line user password is not the same as the WebGUI password.
To clear the WebGUI password get the command line access (SSH or monitor/keyboard)
and open the file:
nano /data/etc/motion.conf
make sure the lines 
# @admin_username admin
# @admin_password
# @normal_username user
# @normal_password
appear as above (nothing after the two password lines).
Save the file, and reboot

NickRudd90, 2021-07-08:
Cheers dude, I will give that a go the moment I get home!!

NickRudd90, 2021-07-08:
All working now thanks for the help

closed 2021-07-08 by NickRudd90

Roobyx, 2023-04-13:
What if you dont have these files in the .conf? I've had a previous installation, then reinstalled it (hence forgot the password) and now nothing changes even if I add these lines. 
I can still see the old username and it still asks for credentials.

starbasessd, 2023-04-13:
It will. In a new install:
The username is "admin" (all lower case)
The password box is blank, unless you had a password when you restored the data.
Select Remember Me 
then login