motioneye-project/motioneyeos #2771: RPI 0 W + MotionEye detection HUB?

icedan, 2021-07-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera

My camera model is: Pi ZeroCam OV5647 5MPx

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

I have a request for help - possibly someone say that it is not possible.
I have several RPI 0 W cameras, as you know, the equipment is not too powerful, but it works on the powerbank for a long time. I also have motioneye in Docker to which I can add cameras.

I have a question, is it possible to do so that RPI 0 W is responsible only for image registration (fast network camera) and motion detection takes place in Docker?
Or some other way to be able to record a smooth image on this RPI 0 at a higher resolution. This image is very smooth on a fast network camera, but RPI 0 is too weak for detection.
If so, what settings?

starbasessd, 2021-07-09:
I run several different versions of motion/motionEye and motionEyeOS as hubs, on I3, i5 (Ubuntu 20.04 & 21.04, and other flavors as needed for testing) VirtualBox VMs (several flavors of linux as needed) and Pi3B+s and Pi4s as needed for testing. I set the PiZeroW or Pi3s as Fast Network Cams, then set the Hub to connect as Network Cameras, or simple MJPeg cameras. All detection and recording is done by the hub. My production machine (I5/16GB/10TB NAS) has currently 6 cameras (3 d-Link PTZ, 2 Foscam PTX, (all on 801.11G network) and a USB Generic Dashcam) and meets my needs, with an average of 10-15 FPS video per camera. YMMV.

starbasessd, 2021-07-09:
Extra: Docker depends on setup. Many issues trying to use Portainer and/or HomeAssistant with the motionEye Docker. A simple hub running Ubuntu or RaspberryPiOS, docker-ce or docker.io, without Portainer or HA work fine. Start adding in other layers and you are on your own...

icedan, 2021-07-09:
Ok, but Simple MJPEG: Adding your device as a simple MJPEG camera instead of as a network camera will improve the framerate, but no motion detection, picture capturing or movie recording will be available for it. The camera must be accessible to both your server and your browser. This type of camera is not compatible with Internet Explorer.

Network should work, will try. Network cameras (or IP cameras) are devices that natively stream RTSP/RTMP or MJPEG videos or plain JPEG images. Consult your device's manual to find out the correct RTSP, RTMP, MJPEG or JPEG URL.

starbasessd, 2021-07-09:
https://github.com/ccrisan/motioneyeos/pull/1764#issuecomment-699571040
Using URL rtsp://ip_address:554/h264
Confirmed working in TCP mode.
You need to go to Video Streaming on the PiZeroW and enable RTSP streaming Protocol and apply.
You were correct about using Simple MJPEG or Remote MotionEye. You can get video, but have no saving or motion detection capabilities for those modes, my apologies.

icedan, 2021-07-09:
> [#1764 (comment)](https://github.com/ccrisan/motioneyeos/pull/1764#issuecomment-699571040)
> Using URL rtsp://ip_address:554/h264
> Confirmed working in TCP mode.
> You need to go to Video Streaming on the PiZeroW and enable RTSP streaming Protocol and apply.
> You were correct about using Simple MJPEG or Remote MotionEye. You can get video, but have no saving or motion detection capabilities for those modes, my apologies.

Thanks. You have 100% right! 4 motioneyeOS on RPI 0 W 1080p30fps. Docker on Qnap TS-473A (AMD v1500). Perfect! Thank you.

starbasessd, 2021-07-09:
Please close if you are up, working, and happy...

icedan, 2021-07-09:
OFC. Thank you again.

closed 2021-07-09 by icedan