motioneye-project/motioneyeos #2772: A Single Raspberry Pi 4 4GB Won't Boot dev20201026

matthew1471, 2021-07-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20201026.

### Board Model

I am using the following board/model: Raspberry Pi 4 Model B Rev 1.4.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: None.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * Raspberry Pi 4 Official Power Supply

### Log Files

I consider the following log files relevant to this issue:

 * None available

### Issue

Logging this for info but I have 6 Raspberry Pis.. on one of them dev20201026 boots REALLY slowy and then crashes and reboots. I have tried the same SD card in other Pis = Fine. I've installed Rasperry Pi OS on a fresh SD = Fine. I've tried. The only thing that fixed it was taking a newer firmware (start4x.elf, fixup4x.dat) and overwriting (start.elf, fixup.dat).. this problematic device then boots:

```
[root@meye-f5803d50 ~]# vcgencmd version
Apr 30 2021 13:46:42
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start_x)
```

So some weird bug must be in the Jul 17 2020 firmware that dev20201026 is bundled with.. that only impacts SOME Raspberry Pi 4 Model B Rev 1.4.

starbasessd, 2021-07-10:
The command you used tells the version of vcgencmd.
The command you need is:
vcgencmd bootloader_version
Which reports for the current stable version:
pi@raspi5:~/pi-gen $ vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1621697261
capabilities 0x0000001f
pi@raspi5:~/pi-gen $ 
Pretty much anything after Sept 2020 should work.

matthew1471, 2021-07-10:
Bootloader is identical on both devices. I left the bootloader alone. This is exclusively fixed by replacing the firmware for some reason..

starbasessd, 2021-07-10:
Works without issue with my Pi4-4GB without grabbing the start4x and fixup4. The issue you will have doing that is the rest of the motionEyeOS won't match, and USB is broken, among other things.

matthew1471, 2021-07-10:
Yep it works fine without grabbing the start4x and fixup4 on 5 devices for me too.. this 6th device it didn't.. I think this might be a firmware bug. I'm going to try grabbing different versions of https://github.com/raspberrypi/firmware/commits so I can tell you which exactly the bug occurs in :)

starbasessd, 2021-07-10:
What firmware? It's a bootloader issue.
What does 
vcgencmd bootloader_version 
report?

starbasessd, 2021-07-10:
BTW, you will find it nearly impossible to apply any of those updates to a Read-Only file system, that is missing almost all the update functionality found in RPiOS...

starbasessd, 2021-07-10:
Showing the differences between the command you ran vs the command you need to run.
pi@raspi1:\~ $ vcgencmd version
May 27 2021 14:02:15
Copyright (c) 2012 Broadcom
version 7d9a298cda813f747b51fe17e1e417e7bf5ca94d (clean) (release) (start)
pi@raspi1:\~ $ vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1624989764
capabilities 0x0000001f

matthew1471, 2021-07-10:
>What firmware? It's a bootloader issue.
It's getting past the bootloader and it's crashing during the kernel.

Raspberry Pi has a bootloader and firmware. The firmware is loaded from start4x and fixup4x, see https://www.raspberrypi.org/documentation/configuration/boot_folder.md

>What does vcgencmd bootloader_version report?
```
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
```
Same as all of them

Yep I am having to plug the SD card into my laptop and modify the 2 files in /boot.

I'll keep testing so I can let you know which specific firmware fixes it.

starbasessd, 2021-07-10:
Try swapping the power supply between the 2, and when running RPiOS run the commands:
vcgencmd get_throttled
vcgencmd measure_temp
If get_throttled reports anything but 0x0 or measure_temp reports anything over 60`C you have issues.

matthew1471, 2021-07-11:
>This is with the identical power supply too.
Already tried the same power supply.. I'm not seeing any throttled.

I am running motioneyeOS on this board and the other 5 cameras. The only thing I am playing with is the firmware files as that seems to alter whether motioneyeOS is bootable on this specific board or not.

I'm excited to find out which firmware bug dev motioneyeOS is hitting.. this should really help us support other people and potentially the issue I saw someone had with a RPi 400

starbasessd, 2021-07-11:
You may want to look at ThingOS and Buildroot for the changes to be applied.
https://github.com/ccrisan/thingos/wiki
https://github.com/buildroot/buildroot
since those are what motionEyeOS is built upon them.

matthew1471, 2021-07-11:
Finished my debugging.

< 17 September 2020 - No Boot
=> 28 September 2020 - Works

https://github.com/raspberrypi/firmware/commit/f0eab3a24f42c06fb6f2360140a00873c2793b1f
f0eab3a24f42c06fb6f2360140a00873c2793b1f

matthew1471, 2021-07-11:
[Updated Pi Boot.zip](https://github.com/ccrisan/motioneyeos/files/6796157/Updated.Pi.Boot.zip)
Here are the files you can replace in /boot (accessible from Windows as D:\ usually as it's a FAT partition) to ensure you're running the latest and greatest firmware.

starbasessd, 2021-07-11:
Created USB boot device using dev20201026 for Pi4
modified cmdline.txt from mmcblk0p2 to sda2
copied your uncompressed zipped /boot files to /boot
copied my wpa_supplicant.conf and ssh.txt files to /boot
ejected and inserted to Pi4-4GB with current bootloader
Boot performance is drastically slower, than same setup (minus your zipped files) for both USB and SDCard on my 2 Pi4-4GB.
Will test on my Pi4-8GB shortly.

starbasessd, 2021-07-11:
Further on 4-4GB, ssh performance is almost nonexistent. top display takes 3-5 seconds, and refreshes are likewise slow/delayed.
Haven't even bothered to install my cameras.

starbasessd, 2021-07-11:
On my Pi4-8GB it does work as normal. Both on bootup, and into ssh session.
Reboot, and system is corrupted. 
Will research further, but I believe it has to do with the USB issues found with other fixup and start replacements.

starbasessd, 2021-07-11:
@ccrisan  I believe this maay be another good reason to run a new build for dev & release.

ccrisan, 2021-07-11:
Will do, as soon as I get some spare time.

klcampbell6502, 2021-09-19:
Just to confirm I'm seeing the same issue with a newly purchased Pi 4 8GB.

starbasessd, 2021-09-19:
You need to update the rpi-eeprom to something from this year (Apr 2021 is latest)

Raspberry Pi Imager provides a GUI for updating the bootloader and selecting the boot mode.

    Download Raspberry Pi Imager

    Select a spare SD card. The contents will get overwritten!

    Launch Raspberry Pi Imager

    Select Misc utility images under Operating System

    Select Bootloader

    Select a boot-mode i.e. SD (recommended), USB or Network.

    Select SD card and then Write

    Boot the Raspberry Pi with the new image and wait for at least 10 seconds.

    The green activity LED will blink with a steady pattern and the HDMI display will be green on success.

    Power off the Raspberry Pi and remove the SD card.

Using raspi-config to update the bootloader

To change the boot-mode or bootloader version from within Raspberry Pi OS run raspi-config

    Update Raspberry Pi OS to get the latest version of the rpi-eeprom package.

    Run sudo raspi-config

    Select Advanced Options

    Select Bootloader Version

    Select Default for factory default settings or Latest for the latest stable bootloader release.

    Reboot

klcampbell6502, 2021-09-20:
> You need to update the rpi-eeprom to something from this year (Apr 2021 is latest)
>
> Raspberry Pi Imager provides a GUI for updating the bootloader and selecting the boot mode.
>
> ```
> Download [...]

I followed the instructions to create and boot from an SD based bootloader, but it made no difference to the problem seen.

I then copied the updated files to my SD card and tried to boot again, but still see the message that "Start4.elf : is not compatible", and am instructed to get a newer version of the software. I note that supplied .zip does not include a replacement for start4.elf, just start.elf.