motioneye-project/motioneyeos #2773: WiFi works, Ethernet doesn't

TonyDong51089, 2021-07-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (dev20201026).

### Board Model

I am using the following board/model: (Raspberry PI 4B 4G).

### Camera

I am using the following type of camera: (MMAL).

My camera model is: (Raspberry CSI).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

A cooling fan
A sd card

### Log Files

Sry, can't access Raspi now.

###Description

Use 2.4G WiFi, it works. Use Ethernet with Raspberry offical OS, it works.
I disabled WiFi in MeyeOs, trying to use Ethernet, now it doesn't show up on router's console.
Router has DHCP enabled.

starbasessd, 2021-07-13:
Clean install of dev20201026 on USB stick for Pi4-4GB (mod cmdline.txt to sda2)
No SSH.txt or wpa_supplicant.conf
insert to Pi, power up
Boots normally, pulls IP address.
Clean install of dev20201026 on USB stick for Pi4-4GB (mod cmdline.txt to sda2)
SSH.txt and wpa_supplicant.conf copied to /boot, eject.
insert to Pi, disconnect etherrnet, power up
Boots normally, pulls IP address for WiFi (priority)
Go to WebGUI. Turn off WiFi network (forces reboot)
re-connect Ethernet during reboot.
Takes an abnormally long time (>30 seconds) to show 'no link'
Takes an abnormally long time for wired network
Eventually shows PING command to router
Setting date & time fails using SNTP
starts up everything else, can ping, but cannot SSH to Pi.
Power off reboot
Boots normally, pulls IP address, able to get into WebGUI and SSH.
Interesting. Will look further.
Suggest work-around, try a power off / on cycle and see if that allows you to connect.

starbasessd, 2021-07-13:
[Issue 2773.zip](https://github.com/ccrisan/motioneyeos/files/6812249/Issue.2773.zip)
Log files for reference.

TonyDong51089, 2021-07-14:
> Clean install of dev20201026 on USB stick for Pi4-4GB (mod cmdline.txt to sda2)
> No SSH.txt or wpa_supplicant.conf
> insert to Pi, power up
> Boots normally, pulls IP address.
> Clean install of [...]

@starbasessd Thanks, very much appreciate it

starbasessd, 2021-07-14:
Are you able to hook up a monitor & keyboard to the Pi?
Do you get a network panic followed by a reboot, or some other error with no wpa_supplicant.conf file and ethernet hooked up?
Try a different port on your router/switch.
Also re-try RaspberryPiOS without a wpa_supplicant.conf file and ethernet hooked up.