motioneye-project/motioneyeos #2774: Optimal setup to maximize FPS in movies.

fujinjr, 2021-07-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911.

### Board Model

I am using the following board/model: Raspberry PI 3B+, Raspberry PI 0w.

### Camera

I am using the following type of camera: Fast Network Camera.

My camera model is: Raspberry Camera PI V2 8 MP.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.
 

I'd like to know if my current setup is optimal in order to maximize FPS in movies.
I attached two images, any opinion? 
Am I using the FNC and RTSP correctly?

[image]
---
[image]

starbasessd, 2021-07-15:
I would upgrade your motionEyeOS to 20200606 or dev20201026, rather than use any 2019 version. There were a number of updates and changes since then.

cap9qd, 2021-07-15:
I would try setting movie pass through on your RPi3B+ instead of omx-h264.
The RPi-0W is encoding the video in hardware to h264 so you shouldn't need
to reencode the stream. Motion daemon on the RPi3B+ will still have to
decode it to do the motion detection but doing pass through might lower
your load. I am not sure this will help you much with fps.

I found if the video was low enough resolution (<1080p) I had better
performance (in terms of CPU load on my motioneyeos server) by leaving my
RPI-0W FNC setup as a mjpeg stream and using the hardware encoder on my
odroid server to encode to h264. I found this recommendation on the motion
project forum that motion deals well with jpegs for the motion detection
algorithm and motion does not hardware accelerate the decode of h264 so the
CPU load takes a hit. Of course I can't find the post now when I need it
but worth a try.

This work was in terms of getting my CPU load down to support more cameras
on my one server so may not be the advice you are looking for here.

starbasessd, 2021-07-15:
@cap9qd Be aware that you lose text overlays (time/date etc) using Pass Through recording, if it is not generated on the camera, which is why I didn't recommend that optimization...

fujinjr, 2021-07-15:
> I would upgrade your motionEyeOS to 20200606 or dev20201026, rather than use any 2019 version. There were a number of updates and changes since then.

I'm not able to physically access the two devices, so I don't want to risk a remote update. 
Anyway is the use of RTSP, in my Pi3 B+ 'Camera Device', correct?

starbasessd, 2021-07-15:
Yes.

fujinjr, 2021-07-15:
> Yes.

I want to clarify some doubts that I have: 
if on the current setup I'll change the Camera Device, on the Pi 3B+, to -> http://192.168.1.112:8081 then I'll should observe a significant drop in FPS, am I right?

What are the disadvantages with my setup... loss in video quality?

cap9qd, 2021-07-15:
Ah yes I am not concerned about text overlay. Sorry for the omission.

starbasessd, 2021-07-15:
Then your setup is almost maxed out. 
@cap9qd 's suggestion to shift to Pass Through is best. To give yourself the best optimization you really do need to update to current versions, just check out the improvements in the change logs here:
https://github.com/ccrisan/motioneyeos/releases

starbasessd, 2021-07-15:
> 
> 
> > Yes.
> 
> I want to clarify some doubts that I have:
> if on the current setup I'll change the Camera Device, on the Pi 3B+, to -> http://192.168.1.112:8081 then I'll should observe a significant drop in FPS, am I right?

Not necessarily, depends on what else is going on. The option slider for framerate limitations isn't absolute. If you have 1 camera, it will try to go that fast, but if the streaming camera is only sending 10 FPS, then that will be your upper limit. If you have other things on your network that eats bandwidth then you may not get the full FPS you've selected to send in FNC mode. On the other hand, if things are pretty idle, you may get the same framerate in both port 8081 vs port554/rtsp...

> 
> What are the disadvantages with my setup... loss in video quality?

fujinjr, 2021-07-15:
> Not necessarily, depends on what else is going on. The option slider for framerate limitations isn't absolute. If you have 1 camera, it will try to go that fast, but if the streaming camera is only [...]

With port 8081 the maximum FPS that I saw was 6 with many lags.

In the following weeks I'll stick with RTSP, I'll increase the 'Image Quality' to 40 on the FNC and see if I'll notice any improvement, quality wise; after all I'm good with 10 stable FPS.

starbasessd, 2021-07-15:
Also, on the PiZero, you will need to watch your bit rate for the stream to get best frame rate, resolution, and quality... PiZero supports 802.11g, but seldom if ever gets anywhere near 54Mbps, which is also dependent on other traffic, the router used, and distance to the router...

fujinjr, 2021-07-16:
Update: with Image Quality up to 40 I don't notice improvements in the image quality; I'll go back to 25. 
I could try to increase the resolution to 720p though.

> Also, on the PiZero, you will need to watch your bit rate...

Take a look:

[image]

fujinjr, 2021-07-16:
Update: with Image Quality set to 40 I don't notice improvements in the image quality; I'll go back to 25. 
I could try to increase the resolution up to 720p though.

> Also, on the PiZero, you will need to watch your bit rate...

Take a look:

[image]

starbasessd, 2021-07-16:
By bit rate I was referring to the PiZero, Settings, Video Device, Bit Rate, currently set to 1Mb/sec. To improve your frame rate & quality, try increasing to 2Mb/sec, and slowly increase until you don't notice any more improvement.  Realize, though, at some point you will saturate your connection to WiFi. This is not brain surgery, just rocket science, where you try a setting and notice a result. Think: SpaceX and Elon Musk....   :)

fujinjr, 2021-07-16:
> try increasing to 2Mb/sec
I did, let's see what happens!

Another question: do you usually activate 'Denoise' option in your FNC cameras?

starbasessd, 2021-07-16:
I activate Auto Noise Detection and Despeckle Filter anytime they are available as an option.

fujinjr, 2021-07-17:
I'm referring to the option inside the pi0:

[image]

starbasessd, 2021-07-17:
Yes, enabled.

fujinjr, 2021-07-18:
I don't notice any difference increasing the bitrate to 2.00 Mbps.
Right now I set the bitrate to 1.25 Mbps and increased FPS to 12 and I get consistent results; one strange thing I noticed is that the Image quality slider (from 25 up to 100), on the Pi0, doesn't affect video quality to my perception.

Question: if I activate RTSP on the Pi0 but then add a camera on the Pi3 using http://192.168.1.112:8081, will it work?

starbasessd, 2021-07-18:
The quickest way to answer your last question is to try it.
The easy answer is Yes.
BTW, the real test of the resolution, quality, frame rate and bitrate changes is not the 'preview' video showing the live feed, but the recorded video.

fujinjr, 2021-07-18:
> BTW, the real test of the resolution, quality, frame rate and bitrate changes is not the 'preview' video showing the live feed, but the recorded video.

I test the video quality, and FPS, with VLC.

starbasessd, 2021-07-18:
BTW, the real test of the resolution, quality, frame rate and bitrate changes is not the 'preview' video showing the live feed, but the recorded video.
VLC is still looking at the preview stream...

fujinjr, 2021-07-18:
> VLC is still looking at the preview stream...

I use VLC with downloaded movies and then I observe Codec information.

Today, in one of the recorded videos, I saw a forward jump in time (of 20 seconds); it looks like someone edited and cut a portion of the video. Is it a possible effect that happens with RTSP?

starbasessd, 2021-07-18:
Yes, occasionally. If/when that happens, you can try adding the camera as UDP instead of TCP.

fujinjr, 2021-07-19:
> If/when that happens, you can try adding the camera as UDP instead of TCP.

How do I do that?

A suggestion for the future: is it possible to add, inside motioneyeos, the possibility to activate a different ISO during the night?

starbasessd, 2021-07-19:
Remove the camera from motionEyeOS
Select the PiZero Camera in the dropdown
Click trashcan icon
Click Add New Camera
Add camera back,
[image]
Select UDP this time.

As to the feature request, please start a new issue with it, so I can tag it as Feature Request.

starbasessd, 2021-07-19:
As an aside, using older versions, I had to use URL: rtsp://192.168.2.120/h264 with the above settings to get UDP to work. There are many issues with the older versions of motionEyeOS and RTSP 9both sending and receiving) that have been fixed in the most recent versions (20200606 and dev20201026) compared to the 2019xxxx versions.

closed 2021-07-19 by starbasessd