motioneye-project/motioneyeos #2775: Feature request: "force trigger" button

snowflakeas, 2021-07-17:
Using MotionEyeOS for a couple of years.

Would really like a Motion "Force Trigger" button (perhaps in the camera view?)

This would allow quick and easy testing of the following, which are important to getting a useful alarm system:
- Test Emailing & drive uploads (and approximate file sizes!)
- Test of video compression settings
- Test of resolution and frame rate

Google groups suggested some command line stuff, but this is very inconvenient for most uses like me that configure MotionEyeOS once every few years (new Pi, system for friends etc).

There is already a similar feature for Still Image capture, but having one that simulates a Motion event would enable much more of the functionality to be tested for each installation in a user-friendly manner, hopefully reducing support requests.

Apologies if this is not the correct place to report - read guidelines, did not see contraindications to feature requests.

starbasessd, 2021-07-17:
The ability to create your own custom buttons already exists:
https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons

starbasessd, 2021-07-17:
Working on some 'quick & dirty' instructions