motioneye-project/motioneyeos #2777: Is this project still being maintained, when's the next release?

sts-ryan-holton, 2021-07-18:
👋 Hi everyone

I thought open an issue here regarding the status of this project, given the number of issues that are being opened and the activity on this project, I'm trying to figure out if and when another release will be made for this project?

The last official release appears to be [the end of October 2020](https://github.com/ccrisan/motioneyeos/releases) and I've been wanting to install and use this project a bit more, but have been hesitant since there hasn't been any new release.

Is this project still being actively maintained, and if so, when can we except a new release?

starbasessd, 2021-07-18:
Yes, it is actively being supported. There are active developers looking at converting to Python3, but it is running on Python2.7 currently. The majority of new issues are camera and video issues. I have most of the RPis to support them. 
I have tried to tag all the issues as they come in (which hadn't been done prior). 
Due to CCrisan not having as many hours available as in the past, he's fallen behind. I do not have a date time frame for the next release, but expect it 'soon'...
Do you have a question, or an issue that needs to be addressed?

sts-ryan-holton, 2021-07-19:
@starbasessd Can you make a recommendation as to what version I should install onto a new Raspberry Pi? I see there's `dev20201026` but has been sitting as a pre-release for quite some time, or should I use the version prior to this?

starbasessd, 2021-07-19:
If it is a Pi4 1GB, 2GB, or 4GB, you need the latest version of the boot eeprom update, and dev20201026
Any other Pi should be able to use 20200606. 
Pi4  8GB may have issues even with dev20201026.

starbasessd, 2021-07-19:
BTW, dev20201026 can boot to USB on the Pi3s and Pi4s with a minor change to /boot/cmdline.txt:
Change /dev/mmcblk0p2 to /dev/sda2

sts-ryan-holton, 2021-07-20:
So `dev20201026` is stable enough for a Pi4 4GB and is recommended? Planning on using it as a Fast Network Camera with a USB webcam transmitting it's signal over Wi-Fi to a nearby Pi4 2GB as a hub for motioneyeos @starbasessd 

Is this release constantly updated then?

starbasessd, 2021-07-20:
Is it stable enough? Yes.
Fast Network Camera only works with CSI (ribbon cable) cameras, not USB.
I would reverse the Pis, if it was possible, to Use the Pi4-2GB as your camera, and the Pi4-4GB as your hub, though. But as I said, you can't use the USB and get Fast Network Camera functionality, it only works with the PiCam CSI type cameras. You can disable everything but streaming, but you still wouldn't have the RTSP stream available. 
As far as being 'constantly updated', what are you looking for? 90% of a desktop or lite OS isn't present in motionEyeOS. You cannot add other applications if you would need to use apt or dpkg, as they and their support applications are not present. You CAN add python and bash scripts, some drivers (if they can be installed via a copied folder), and such. but it is NOT a full OS, just enough OS to turn it into a NVR.

sts-ryan-holton, 2021-07-20:
@starbasessd Ah, right, because that's why I ordered a Pi4 with a USB camera because the regular PiCam isn't very good. Does this mean I've effectively wasted money in buying a USB camera in the hope that plugging it in would work? That doesn't seem right, I've seen people using USB cameras on their Pi with Motion Eye OS?

starbasessd, 2021-07-20:
Yes, you can use USB cameras with little to no issues, as long as it works with standard RaspberryPiOS, or otherwise doesn't need a driver and is supported by V4L2. I use several different USB cameras all of which are supported by V4L2, including a generic TIGA Dashcam, and generic Logitech webcam. 
Fast Network Camera is a feature of MotionEyeOS, and only CSI cameras are supported as stated in the wiki page:
https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera

sts-ryan-holton, 2021-07-20:
I suppose I'll have to use something like Raspivid for this Pi to create a feed that can be read by the Pi4 2GB hub? That would work presumably, the hub could read any IP address camera and save the video?

starbasessd, 2021-07-20:
Beyond the scope of my support, sorry.

starbasessd, 2022-01-19:
No response in more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd