motioneye-project/motioneyeos #2779: Schedule different ISO at different hours

fujinjr, 2021-07-19:
Is it possible to add, inside motioneyeos, the possibility to choose a different ISO during the night?

starbasessd, 2021-07-19:
Flagged

sts-ryan-holton, 2021-07-19:
This is a feature I've wanted for a while, it would be nice to have some kind of "go-to" wiki solution if this can't be achieved in code, I actually think it would be great to control other settings as well for a "schedule"

starbasessd, 2021-07-19:
You can do it many different ways in scripting.
The easiest (IMO)
install camera.
copy /data/etc/camera_X.conf to /data/etc/camera_X.conf.primary
make changes
copy /data/etc/camera_X.conf to /data/etc/camera_X.conf.secondary
create a script called /data/etc/daytime:
- - - - -
\#!/bin/bash
cp /data/etc/camera_X.conf.primary /data/etc/camera_X.conf
service motioneye restart
- - - - -
create a script called /data/etc/nighttime:
- - - - -
\#!/bin/bash
cp /data/etc/camera_X.conf.secondary /data/etc/camera_X.conf
service motioneye restart
- - - - - 
edit crontab
crontab -e  \# select nano
- - - - -
add line:
\* 6,18 \* \* \* service motioneye restart \# this will restart motioneye at 6a and 6p set as wanted
\<ctrl\>x, y, \<enter\>
- - - - -
There are many other ways to do it...

Ventnts, 2021-09-21:
I have 3 kids and the light switch doesnt really cut it somtimes, even though I love how Cris took the time to figure out cool stuff like that.

I agree, I think the schedule manager could use some TLC.

starbasessd, 2021-09-21:
See my response to your other ticket.