motioneye-project/motioneyeos #2780: Python script to show Power / Heat issues on RPis

starbasessd, 2021-07-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Applies to most

### Board Model

I am using the following board/model: Applies to most Raspberry Pis

### Camera

I am using the following type of camera: Any supported

### Network Connection

My motionEyeOS unit is connected to the network via: Applies to any

### Peripherals

I am using the following peripherals that I consider relevant to this issue: N/A

### Log Files

I consider the following log files relevant to this issue:

Here's a handy little script to show Heat and/or Power issues:
Create a file 
nano /usr/bin/oops.py
paste in the following:
- - - - -
#!/usr/bin/env python2

import subprocess

GET_THROTTLED_CMD = 'vcgencmd get_throttled'
MESSAGES = {
    0: 'Under-voltage!',
    1: 'ARM frequency capped!',
    2: 'Currently throttled!',
    3: 'Soft temperature limit active',
    16: 'Under-voltage has occurred since last reboot.',
    17: 'Throttling has occurred since last reboot.',
    18: 'ARM frequency capped has occurred since last reboot.',
    19: 'Soft temperature limit has occurred'
}

print("Checking for throttling issues since last reboot...")

throttled_output = subprocess.check_output(GET_THROTTLED_CMD, shell=True)
throttled_binary = bin(int(throttled_output.split('=')[1], 0))

warnings = 0
for position, message in MESSAGES.iteritems():
    # Check for the binary digits to be "on" for each warning message
    if len(throttled_binary) > position and throttled_binary[0 - position - 1] == '1':
        print(message)
        warnings += 1

if warnings == 0:
    print("Looking good!")
else:
    print("Houston, we may have a problem!")

- - - - -
The file is also attached
run command:
chmod +x /usr/bin/oops.py
You can run it as oops.py
[oops.zip](https://github.com/ccrisan/motioneyeos/files/6849225/oops.zip)

pir8s, 2022-02-10:
Hello, thanks for this script.
Would it be possible for you to post an updated version of this script so it works in Raspberry pi bullseye ?
python2 is no longer included in bullseye but 3.x and all I get is an error stating:
"/usr/bin/env: ‘python2’: No such file or directory"

Thanks

starbasessd, 2022-02-10:
Not for a while, as the line:
> throttled_binary = bin(int(throttled_output.split('=')[1], 0))

doesn't convert nicely. You can wait, or install python2-minimal which is available, and since python2 is what motionEye (on another OS) and motionEyeOS uses, would be installed, anyway. If you are trying to use the script outside motionEye or motionEyeOS, then I suggest asking the original author at:
https://harlemsquirrel.github.io/shell/2019/01/05/monitoring-raspberry-pi-power-and-thermal-issues.html