motioneye-project/motioneyeos #2781: USB camera video resolution isn't updating

sts-ryan-holton, 2021-07-21:
### motionEyeOS Version

- dev20201026

### Board Model

- Raspberry Pi 4, 4GB model

### Camera

I'm using a USB webcam, from the brand Benewy

### Network Connection

I'm connected over Wi-Fi

### Peripherals

Just the USB camera

### The problem

I'm currently trying to adjust the video resolution of my camera under the Video Device section, however, when setting the resolution to a larger resolution and saving, the new resolution does not change, it's always what appears to be 640x480.

The USB webcam I've got supports resolutions up to at least 1280x720, so that's what I'd like,.

I did try setting: `height 800` as a test in the Extra Motion Options, but this appeared to crash the system.

What am I missing and how can I set the resolution of this camera?

[image]

starbasessd, 2021-07-21:
While I am trying to throw together a Pi4-4GB dev20201026 with USB camera, please try lowering your framerate to 15 or 10 FPS for testing purposes.

sts-ryan-holton, 2021-07-21:
I've just lowered both the video streaming and main video to 10fps, and stream quality to 75%. If it helps, here's some other version info:

- Motion Version: 4.3.1
- motionEye Version: 0.42.1

[image]
[image]

starbasessd, 2021-07-21:
Hmmm. 
Do you click Apply at the top of the window?
Do you have 800x600 in your dropdown for resolution? 
Can you select Custom and try 1024x768 and 800x600?
I do not get the Brightness, Contrast and Hue options on my Tiga
I also set Fit Frame Vertical and 1 column options in Settings, Preferences..
If you can get it to record video, in VLC you can view framerate and resolution (and other parameters), what does it record as?
Have you connected your other Pi4 to it (the Pi4 with the camera) as Network Camera
and tested?
Which make and model of USB camera? 
If I go to Benewy.net and check products, webcam they have nothing listed and their support is a hotmail account.

sts-ryan-holton, 2021-07-21:
I'll have to look into the fit frames vertically option and the others as you described, I'm currently not at the Pi so will clarify those questions as soon as I can, but what I can check here is:

- I do click "Apply" at the top of the screen, the Pi then does it's spinning and then reloads.

- I've tried setting "Custom" as a resolution to 1280 width and 720 height, this made no difference but I can try the exact measurements

- the camera is [this one](https://www.amazon.co.uk/Microphone-Desktop-Conference-Studying-Compatible/dp/B089LTY8X7/ref=mp_s_a_1_1_sspa?dchild=1&keywords=benewy+webcam&qid=1626897356&sprefix=Benewy&sr=8-1-spons&psc=1&smid=A3J9ITXBRZ4BLV&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExUjExUFhSNUNSSENRJmVuY3J5cHRlZElkPUEwNzA1MTkwM0Y0SExDTUpSUTlKMSZlbmNyeXB0ZWRBZElkPUEwNTc5MTk3VVlFTEJIUUlaN1o1JndpZGdldE5hbWU9c3BfcGhvbmVfc2VhcmNoX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=)

starbasessd, 2021-07-21:
Curiosity: do you have any other USB devices plugged in?

starbasessd, 2021-07-21:
Also, if-when you can SSH or get to the console, run the command:
vcgencmd get_throttled
and post result, please.

sts-ryan-holton, 2021-07-22:
Here's some more findings:

1. Running: `vcgencmd get_throttled` gives me the following: `throttled=0x0`.
2. I don't have any other USB devices plugged in, just the camera plugged into a USB 2.0 port, I've tried both ports and also both USB 3.0 ports with the same results.
3. I do have `800x600`, I've just tried this and the resolution doesn't change from `640x480`.
4. Weirdly, if I set a smaller resolution, like `320x240` this resolution works, but changing it back to the resolution I desire just reverts it back to `640x480`.
5. Setting `1024x768` and `800x600` with the Custom option doesn't set these as the resolution, resolution remains unchanged.

starbasessd, 2021-07-22:
Interesting. I am at a loss diagnosing in motionEyeOS. Do you have another SDCard or USB Thumb drive to load RaspberryPiOS to run some other commands?

starbasessd, 2021-07-22:
lsusb doesn't provide info that would help diagnose, and most other tools are missing, too...

sts-ryan-holton, 2021-07-23:
Maybe the Extra Motion Options would allow me to set the height and width? What format would I need to put in here to try that?

starbasessd, 2021-07-23:
One option per line
same format as motion, as found here:
https://motion-project.github.io/motion_config.html#width
the whole document is helpful
https://motion-project.github.io/motion_config.html

sts-ryan-holton, 2021-07-23:
Okay, so I just tried this in the extra motion options:

```
height 1280
width 720
```

And I get an error:

> An error occurred. Refreshing is recommended.

And after refreshing it says I haven't configured any cameras, so rebooting the pi kind of fixes the config, but why is this broken?

sts-ryan-holton, 2021-07-23:
My logs:

[motion.log](https://github.com/ccrisan/motioneyeos/files/6870289/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/6870290/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/6870291/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/6870292/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6870293/motioneye.log)

sts-ryan-holton, 2021-07-23:
Okay, so I'm noticing in the log file when I attempt to change the video resolution these kinds of warnings?

> Jul 23 18:47:06 meye-carcam user.warn motion: [1:ml1:Car Cam] [WRN] [VID] v4l2_pixfmt_set: Adjusting resolution from 1280x720 to 640x480.

This seems to be in line with what I'm experiencing, each time I try and set a larger resolution, the video always appears as `640x480`, but in the dropdown it's still set to the resolution I desire, what does this mean?

sts-ryan-holton, 2021-07-23:
Right, somehow I've managed to get it working after some more digging, so I came across this command online that apparently reads the supported resolutions of a camera:

- `v4l2-ctl --list-formats-ext`

The output gave me:

[image]

And then I came across the [v4l2_palette](https://motion-project.github.io/motion_config.html#v4l2_palette) option, and then it occured to me that the default (number 17) which is reading from `YU12` adds up with the very few resolutions in my output above, but MJPG has more resolutions including the one I need, so in the extra motion options I added:

- `v4l2_palette 8`

Applied the changes and now my resolution has been set successfully!

closed 2021-07-23 by sts-ryan-holton