motioneye-project/motioneyeos #2782: Network camera shows green/grey screen when adding to a MotioneyeOS hub

sts-ryan-holton, 2021-07-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

- dev20201026

### Board Model

- Raspberry PI 4B 4GB

### Camera

- I'm using a USB webcam, from the brand Benewy, [this one](https://www.amazon.co.uk/Microphone-Desktop-Conference-Studying-Compatible/dp/B089LTY8X7/ref=mp_s_a_1_1_sspa?dchild=1&keywords=benewy+webcam&qid=1626897356&sprefix=Benewy&sr=8-1-spons&psc=1&smid=A3J9ITXBRZ4BLV&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExUjExUFhSNUNSSENRJmVuY3J5cHRlZElkPUEwNzA1MTkwM0Y0SExDTUpSUTlKMSZlbmNyeXB0ZWRBZElkPUEwNTc5MTk3VVlFTEJIUUlaN1o1JndpZGdldE5hbWU9c3BfcGhvbmVfc2VhcmNoX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=)

### Network Connection

- Wi-Fi

### Peripherals

- Just a USB camera

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/6870606/boot.1.log)
[dmesg).log](https://github.com/ccrisan/motioneyeos/files/6870607/dmesg.1.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/6870608/messages.3.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/6870609/motion.2.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/6870610/motioneye.2.log)

### Problem

I'm trying to link the video streaming feed from my Raspberry Pi 4 (with USB webcam) to my central Raspberry Pi 4 hub (which currently already has a Pi Zero with the Pi Cam fed into this hub, which will soon be replaced) but when I try to add a network camera, the feed is either shown as green or grey.

My streaming URL from my Pi 4 with the webcam is: http://192.168.1.147:8081 and when adding this as a network camera to my hub, I'm not seeing any feed of the video.

If I choose a Simple MJPEG Camera as the option for adding a camera then the feed is visible, but I loose the option to record.

Here's what the configuration looks like when I add my camera:

[image]

When I do this, my existing Pi Zero feed breaks (Camera1), but more importantly, I get a green screen for the network camera (Pi 4 with USB camera feed) that I try to add, why?

[image]

What am I missing?

starbasessd, 2021-07-23:
Username and or password? Even if username is admin or user, and the password is blank, you still need them there.

sts-ryan-holton, 2021-07-24:
Okay so I set a password and username for the Surveillance credentials on my Pi4 with webcam, for the purposes of this issue, I set it to **user** and **password**. I then plugged the following data into the Network Camera option on my Pi4 hub and added the camera, initially it was a green screen, and after about 20 seconds turned grey and displays:

> Connection to camera lost

I can still access the feed of my camera on it's IP with the port http://192.168.1.147:8081/

[image]

starbasessd, 2021-07-24:
There was no need to set a password.
Does the URL work for a web browser? Or VLC? Can you ping the IP address? I know you said you were able to get a Simple MJPeg stream to work, but something doesn't add up.

sts-ryan-holton, 2021-07-24:
I can ping the IP address, the URL does work fine in a web browser and I can view the video stream in VLC media player when I click on Add Network Stream and enter: http://192.168.1.147:8081/

starbasessd, 2021-07-24:
On the Pi with the camera: note the resolution set in Video Device.
On the Pi you are trying to add the Network Camera, in Video Device, be sure to use the same resolution.

sts-ryan-holton, 2021-07-24:
On the Pi I'm trying to add the camera to, there's no resolution option in the Video Device settings, here's a comparison:

Pi with Camera

[image]

Pi used as Hub (where I'm trying to add camera)

[image]

starbasessd, 2021-07-24:
You have me at a total loss. I'll leave it here to see if anyone else can assist.

sts-ryan-holton, 2021-07-24:
Hopefully someone else will be able to contribute to this soon since this is quite a major thing

sts-ryan-holton, 2021-07-25:
@starbasessd Today I tried adding my Pi 4 with webcam to my pi hub as a Remote MotioneyeOS Camera, on port 80. this adds to the hub almost instantly and shows the stream, much like the MJPEG option, however, it appears that the option I need is the Network Camera since I can't update the config independently of the camera on my hub when added via the Remote MotioneyeOS Camera type (it simply just seems to fetch the config remotely), and also, the MJPEG option doesn't give me any recording options.

I need the Network Camera type, but each time I add this I get the green screen, followed by a grey screen with the message "Unable to open video device".

I did try removing one of the cameras from my hub, but has to painstakingly add the config for this camera back after realising that there's no difference with just one cam

sts-ryan-holton, 2021-08-22:
Any update on this?

starbasessd, 2021-08-22:
If you add the Pi4 WebCam As Network, URL: rtsp://ip_address:554
Username admin
password (blank or whatever you set it to)
try both RTSP/TCP or RTSP/UDP for camera
What results do you get?

gabrielconlon, 2021-10-22:
Any resolution on this?  I am having a similar issue.  using a RPI-4 with USB camera and a RPI0w with IR camera on the board.  I've tried both directions of link to no avail, always getting a grey screen from the networked camera

MitchDresdner, 2021-11-14:
bingo!

instead of http://...
try: 
   mjpeg://{yourip}:{port}

i entered the userid per starbasessd

solved my same issue as reported above.

had to read the fm 
  https://motion-project.github.io/4.1/motion_guide.html#netcam_url

lukasdoerr, 2022-03-21:
> bingo!
> 
> instead of http://...
> try:
> mjpeg://{yourip}:{port}

Its a little late bit late, but i can confirm, that fixes the green image problem! I can reproduce this procedere on multiple devices. Every device works now. @starbasessd can you mark @MitchDresdner's Answer please?