motioneye-project/motioneyeos #2783: Network Share preserve movies for one day deletes everything in /

sts-ryan-holton, 2021-08-01:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

```
motionEye Version 0.42.1
Motion Version 4.3.1
OS Version motionEyeOS dev20201026
```

### Board Model

- Raspberry PI 4 (4GB)

### Camera

- An attached USB webcam

### Network Connection

- WiFi

### Peripherals

- Just the USB camera

### The problem

I've currently got my Raspberry Pi 4 set to record movies continuously to a Network Share, which in my context is a Synology NAS DS215J enclosure with 2 drives.

On this NAS, I've got a sharename set up called `motioneye`, this is a shared folder on the Synology.

I've plugged all of this into the Motioneye OS file storage tab, which I'll attach a configuration to.

My issue is that when I got to set my movies to delete daily, it says that it will delete everything inside of `/`, and I'm not sure whether this is everything on my Synology, or just everything in the sharename?

> This will recursively remove all old media files present in the directory "/", not just those created by motionEye!

How can I be sure that only those files inside of the share will be deleted, and when the notice says it will remove all old media files, what command is this running? Will it just delete MP4 files?

[image]

starbasessd, 2021-08-01:
As long as there isn't anything else being stored in synology/motioneye, nothing to worry about. 
The share is not the root of the synology, I hope. IOW if you go to synology/ you should be able to see all the folders on it, but if you go to synology/motioneye, you should only see what is in that folder tree. It considers synology/motioneye to be the root. You could add /cameras as root, if you wanted, and it would consider synology/motioneye/cameras to be the base of the tree.
Recursively deleting would be going synology/motioneye/cameras/date/hour/files, not synology/ and ALL the folders available...
As to the command used, it's buried in a compiled python script, and I'll see what I can find for you.

starbasessd, 2021-08-01:
This should be the same as what's in motionEyeOS as far as deleting media files:
https://github.com/ccrisan/motioneye/blob/dev/motioneye/mediafiles.py
Using python calls to delete the files.

sts-ryan-holton, 2021-08-01:
So is `/` the root of the share, or of the entire NAS?

starbasessd, 2021-08-01:
/ is the root of the share, not the whole NAS (unless you oopsed and pointed the motioneye share on the NAS to the root of the NAS, it happens) which is why you should connect to the share from a PC to confirm what you see...