motioneye-project/motioneyeos #2784: Odroid HC1 reboot does not work

Klausoleum, 2021-08-07:
Hello, 

I tried an update on my Odroid HC1 (server without cam). I used the download version of the Odroid XU4 for this. The hardware is the same (HC1 has no HDMI) and has always worked so far.
All versions up to and including 11 Sep 2019 (20190911) work perfectly.

With all versions that came after that, the reboot generally does not work. If the power supply is interrupted, the system will restart properly.

Can it be related to the kernel version? From version 20200203, which no longer works on the HC1, the kernel of the XU4 was changed to version 4.14.
Unfortunately, the HC1 is now also discontinued.

Does anyone know where my problem is?

starbasessd, 2021-08-07:
Did you look at:
https://github.com/ccrisan/motioneyeos/issues/908
https://github.com/ccrisan/motioneyeos/issues/1892

Klausoleum, 2021-08-07:
> 
> 
> Did you look at:
> #908
> #1892

Thanks for the referrals @starbasessd .
My system boots great. Only the reboot does not work. After the reboot, only the red LED lights up on the board. The blue LED no longer flashes.
After an interruption in the power supply, everything starts up again.

starbasessd, 2021-08-07:
Which is the same symptom on some of the other odroids. Works fine first boot, then reboot goes south. Usually because motionEyeOS tries to /extend/format the partitions, but the system oops's. Does yours have emmc?

starbasessd, 2021-08-07:
Do you have a USB Thumb or other drive attached?

cap9qd, 2021-08-07:
If you change the boot options I will reboot but will puke out quite a bit
of kernel info.

Also there are instructions for getting the serial consol running if you're
interested.

https://github.com/ccrisan/motioneyeos/issues/2597#issuecomment-766144660

cap9qd, 2021-08-07:
Here is the serial console changes.

https://github.com/ccrisan/motioneyeos/issues/2655#issue-792615486

Klausoleum, 2021-08-07:
> 
> 
> Which is the same symptom on some of the other odroids. Works fine first boot, then reboot goes south. Usually because motionEyeOS tries to /extend/format the partitions, but the system oops's. Does yours have emmc?

i don't use emc.
There is only one HDD installed on the SATA port. otherwise nothing is plugged in or connected (except ethernet).

cap9qd, 2021-08-17:
We're you able to implement the changes for the reboot fix? Just curious if
my observations and changes can br confirmed by someone else with an HC1/2.

starbasessd, 2021-08-17:
How does your drive show up? mmcblk0p1 or sda1 or something else?

Klausoleum, 2021-08-17:
> 
> 
> How does your drive show up? mmcblk0p1 or sda1 or something else?

the hard drive is mounted as sda1

[image]

Klausoleum, 2021-08-17:
> 
> We're you able to implement the changes for the reboot fix? Just curious if my observations and changes can br confirmed by someone else with an HC1/2.
> […](#)

yes, I can do it with pleasure. Can you quickly link the reboot to me?

I'm currently back to a working 2019 version. The 2021 version also seems to have a bug when saving videos. after a while, only defective files with 0 or 269 bytes are saved.

If necessary, I can run tests on a different memory card at any time.

starbasessd, 2021-08-17:
@Klausoleum Do you have a file in /boot like cmdline.txt? If you do, is there something like root=/dev/mmcblk0p2 or 
root=/dev/sda2 
there? 
If so, can you change it to root=/dev/mmcblk1p2
I'm wondering if either of the 1st two are hard coded, and needs to be 'fixed' to a floating value.

Klausoleum, 2021-08-19:
@starbasessd 
I found the information in the boot.ini.
The right information is already there, just as you wanted to change it.

`ODROIDXU-UBOOT-CONFIG
setenv fdt_high "0xffffffff"setenv bootcmd "fatload mmc 0:1 0x40008000 zImage; fatload mmc 0:1 0x42000000 uInitrd; fatload mmc 0:1 0x44000000 exynos5422-odroidxu4.dtb; bootz 0x40008000 - 0x44000000"
setenv bootargs "console=tty1 root=/dev/mmcblk1p2 rootwait ro no_console_suspend panic=10 quiet loglevel=1"
boot`

starbasessd, 2021-08-19:
Sorry, it was a shot...