motioneye-project/motioneyeos #2786: configured as Wifi AP, ap0: ERROR no device found

k8md, 2021-08-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi3-20200606.

### Board Model

I am using the following board/model: Raspberry PI 3A+.

### Camera

I am using the following type of camera: MMAL.

My camera model is: Raspberry Pi Camera Module V2.

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi AP
[config files.zip](https://github.com/ccrisan/motioneyeos/files/6972503/config.files.zip)
.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: none

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
I followed the instructions to set up motioneyeos as an AP.  The first boot up, it works fine.  I am able to connect and see the camera.  After power down and a reboot, the wifi is never started.  I attached a monitor and saw the following during bootup:

Starting dnsmasq: SIOCSIFADDR: No Such Device
ap0: ERROR while getting interface flags: No Such Device
done

starbasessd, 2021-08-12:
Also as an option, could you try
https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz
with your currently attached version of the files?

starbasessd, 2021-08-12:
I did test a 3B+ with the 20200606 image and the ap0 files and it worked. I don't know if its a reduced RAM issue or a different WiFi chip, which is why I'd like you to test the old file set (uses wlan0 instead of ap0) and the dev20201026 image. Please let me know...

k8md, 2021-08-12:
using 20200606 and wlan0 works!  20201026 and ap0 did not work.  20201026 did the same thing.  It works fine on the initial boot, but does not work on subsequent boots.

k8md, 2021-08-12:
Just tried 20201026 with wlan0 in the configs and it's working fine.