motioneye-project/motioneyeos #2787: Streaming on demand & full resolution still pictures

maxbcn, 2021-08-16:
Preliminary Docs

I confirm that I have read the CONTRIBUTING guide before opening this issue.

I confirm that I have read the FAQ before opening this issue.

motionEyeOS Version:

motioneyeos-raspberrypi4-dev20201026

Board Model

    Raspberry PI 4 model B

Camera

    CSI HQ Raspi-cam 12 Mpx

Network Connection

Ethernet

The problem / question:

Q 1 : Hello, I am using motioneye with Raspberry Pi out of my network/ @ office. Is a 300Km away.
 Is the streaming on all the time? Just because is with a router with a sim card 15Gb with static IP, and I would like to know if is going to spend all the data in 3 days or the streaming can be just when someone is trying to conet to it ? The system is powered with solar panels ,& batteries , and a clock than turns on and off every day ( on 8am - off 18 pm )

For me would be good to be on demand, so only spend data  from the sim card, and power ,  when someone whants to acces to view the streaming.

My setting are, still Image every 600 seconds. 

Q 2 : My camera gives  4056 x 3040 pixels (12 Mpx ), but I only want to streaming and still photo , no motion detection and no recording video at all. How can I get full frame pictures from the camera with motioneye still images to do timelapse full frame ?

Q 3 : Is there a way to see the Raspi interface ? maybe to install Teamviewer, Change cron parameters,.... etc    . If so How to do that. Or SH way ?
I try that , but not allow me to go thru it .

Thank You 

Marcos

maxbcn, 2021-08-16:
Thank you for your quick answer!!!!

*YOU SAID """A1) Sort of. It doesn't 'push' data, unless you are doing notifications. It only 'streams' data when someone is connected.
As to only working when someone is trying to connect, that would have to be an outside the Pi device that senses the request for connection (like 'wake on lan').
If you are taking a snapshot every 10 minutes (600 seconds) you don't want your Pi constantly shutting down and starting up.""""**

I keep the raspberry up all the time , from 8am to 18 pm, but  if someone, authorized, wants to see the streaming just puts the IP + port  and they will se the streaming, and @ the same time the system is taken a picture every 600 seconds, during this period.

Normally I put in the cron a timer to disconnect, turn off the RPI,  every day @ 18:05 , so the RPI don´t get damage with on- off all the time. Netx morning all start again whith a extrenal clock attached to the battery system. So the RPI turns on.

Is there a way to shut down the RPI ? or just unplug it.

Thank you. 
Marcos

bbillp, 2021-08-17:
To turn the Pi off and on and with power management check out the WittyPi board at adafruit.com or the robotshop.com.     Its a programmable top hat board with real time clock and power management.   You can setup any number of off/on sequences by time and date , I have 4 of these boards on pi 1-2-3 and 4 boards for the allsky camera and other projects.     Bill P.

starbasessd, 2021-08-17:
By software command(s):
sudo shutdown -h now
or
sudo halt
Wait about 10-30 seconds (visually, the green activity light will pulse 10 times then it's safe)
with a physical switch, instructions here:
https://magpi.raspberrypi.org/articles/off-switch-raspberry-pi
Never a good idea to 'just kill power' especially when using a micro SDCard as you don't know if you are interrupting an important file write operation.

maxbcn, 2021-08-17:
Hello all, 
bbillp: With the motioneyesOS , I dont have acces to terminal through ssh ,  to configure  your option  ( wittyPI board ) As I know. So I can not configure anything regarding inside the OS . I am right or wrong ? If I can , tell me how. I try to get through ssh as yo can see in the picture , but it keep asking me for a password, and I have tried different ones and not way to get through. ( permission denied ( publickey, password, keyboard-interactive))

[image]

Thank You

starbasessd, 2021-08-17:
User name is not pi, it's admin. If you did not set a password in the WebGUI, it's blank.

starbasessd, 2021-08-17:
You can create python and bash scripts in motioneyeos, but cannot install applications using tools like 'apt'.

maxbcn, 2021-08-17:
So I can have acces to cron from it ? So I can create a shut down the RPI

Thank you 
Marcos

starbasessd, 2021-08-17:
Correct. Another thing to be aware of: 
When signed in as admin, you ARE root, no need to use 'sudo' for commands. 
Editing cron is as simple as:
crontab -e
add your lines
\[ctrl\]x y \[enter\] to save and enable
Most of the OS is read-only
you can make changes to things in /data/... (like /data/etc/motioneye.conf)

maxbcn, 2021-08-17:
I am in through ssh , thank you . crontab -e gives me a blanc folder with nothing on it, su I just put the job to shutdown.

I try a cron job : 01 14 * * * shutdown -Ph now
Will that shutdown the RPI ?

starbasessd, 2021-08-17:
I believe it should be:
01 14 * * * /usr/bin/shutdown -Ph now
to have it shut down at 2:01p
You could then have your power switch timer turn it off safely, then turn it on in the morning.

maxbcn, 2021-08-17:
Hello again, The job to shutdown by cron that you tell me , just don´t work .
I have tried diferents ones:

04 20 * * * admin shutdown -Ph now
06 20 * * * shutdown -Ph now
08 20 * * * admin /usr/bin/shutdown -Ph now 
 11 20 * * *  /usr/bin/shutdown -Ph now 

No one do the job. Do you think it should be a different one?

Thank you again

Marcos

starbasessd, 2021-08-17:
shutdown in any flavor doesn't exist in motionEyeOS.
The equivalent is: 
halt -d XXX #with XXX being the number of seconds delay
or
poweroff -d XXX #with XXX being the number of seconds delay
so the command in crontab would be:
XX YY * * * /sbin/halt -30 
or 
XX YY * * * /sbin/poweroff -30
for a 30 second delay at YY:XX time

maxbcn, 2021-08-17:
Hello
XX YY * * * /sbin/halt   it works SUPER!!!!

Thanks to everyone who has helped me with this.
Marcos

starbasessd, 2021-08-17:
after confirming for a day or 2 don't forget to close the issue.

maxbcn, 2021-08-17:
Will do , Thank you

maxbcn, 2021-08-18:
All goes fine !! Thank You

closed 2021-08-18 by maxbcn