motioneye-project/motioneyeos #2791: Will FNC recording ever be an option?

d-looper, 2021-08-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: Camera is set to MMAL, but question is about MJPG

My camera model is: Raspberry Pi Zero Camera Module 160 FOV 5MP Fisheye Lens Camera

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

I have a general question. 

Will recording ever be possible using the Fast Network Camera option?

starbasessd, 2021-08-20:
The purpose of FNC is to use camera only funtionality to stream a higher rate/quality, and to achieve that end, disables all motion detection and recording.
https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera

d-looper, 2021-08-20:
That was fast. Yeah, i read the wiki on it. I guess i was was just hoping that at some point it would be possible to use that functionality on the camera but still be able to record at the MotionEye hub for smoother recordings.

starbasessd, 2021-08-20:
You can turn on Passthrough on the Hub to help, but its a matter of trading priorities. The Pis are limited in processing power, the PiZero even more so (single core vs quad core). You could move the camera to a Pi3 or Pi4 and use a different Pi3 or 4 for your hub with better results.

d-looper, 2021-08-20:
The hub is an HP server running Ubuntu, no shortage of processing power there. I get the low processing power on the zero, but, if the hub (server) is where the recording is happening, wouldnt that take the processing needed to record from the pi? Essentially the pi would only be streaming the video while the hub is doing all the recording work. Or do i just have this all wrong?

starbasessd, 2021-08-20:
I am confused, then. If you have the hub doing the 'heavy lifting', why do you need recording on the PiZero?

d-looper, 2021-08-20:
Sorry, let me explain. 
I am planning on a 6+ camera setup. Each camera will be using a Pi Zero running MotionEyeOS. The cameras will be connected to the MotionEye hub i have installed and setup on an Ubuntu server. I would prefer to have "Fast Network Camera" turned on for each camera as the framerates are soooo much better. However, if i do that, i can no longer record the streams from the cameras. That's where my question comes in. 

Does that make more sense?

starbasessd, 2021-08-20:
OK, let me throw up a VM Ubuntu and a PiZero PiCam, and I'll get back with you in a bit.
One more question: How are you adding the PiZero FNC to the Server Hub? Specifics, please.

d-looper, 2021-08-20:
OK, so i apparently all i had to do was talk to you to make it work... I re-added as a network camera and now its working.. I tried this multiple times last night and it wouldnt find the camera/feed.

starbasessd, 2021-08-20:
I just loaded a PiZero with a PiCam (CSI ribbon cable) and dev20201026 (already on the card from a previous test) Set to Fast Network Camera
I connected to a Ubuntu 20.04 LTS with motion 4.2.2, motionEye 0.41
Added PiZero as Network Camera, URL http://ip_address:8081, user admin, password blank
Was able to set up Movies with both motion triggered and continuous recording.

starbasessd, 2021-08-20:
I also threw some pixie dust at you, and invoked Linus Torvald's name...

d-looper, 2021-08-20:
For some reason it wasnt working for me that way until after i submitted this question. Thank you for your time, its working great now. You guys are doing an awesome job!

closed 2021-08-20 by d-looper