motioneye-project/motioneyeos #2794: wired connection panic

kner7, 2021-08-23:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606.

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: no cam at the moment

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

I consider the following log files relevant to this issue:

 cant find logs

I cant get past the first step. Pi boots and shows the first couple procedures then hangs at configuring wired connection then panics and reboots.

I've also tried wifi with conf file but it fails that one too. I need it to work with ethernet and i cant find anyone that had a similar issue and fixed it. Anyone an idea?

starbasessd, 2021-08-23:
If you put a Raspbian Lite on the SDCard, does it boot and get an IP address? 
Do you get either or both Ethernet lights?
Please try dev20201026 https://github.com/ccrisan/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi3-dev20201026.img.xz
If you want to try WiFi, then the following (with changes to SSID, Password (psk) and country as needed) is the format needed:

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="SSID-1"
    psk="p@$$W0RD"
	key_mgmt=WPA-PSK
}
network={
    ssid="SSID-2"
    psk="P@$$w0rd"
	key_mgmt=WPA-PSK
}

Needs to be created by a linux friendly editor: Notepad++ (not Notepad), nano, vi, etc...

kner7, 2021-08-23:
just tried dev20201026, still getting the same problem, wifi with your format also still says failed. I'm getting one ethernet light on the pi3.

starbasessd, 2021-08-23:
What about RaspberryPiOS Lite?

kner7, 2021-08-23:
yes, it has an ip on raspberryPiOS just booted it

starbasessd, 2021-08-23:
I don't know. I have 2 Pi3s here a 3B and a 3B+, both using SDCard, both using Ethernet, and both booted with IP addresses, with 20200606 and dev20201026. No wifi. Do you add anything else to /boot before 1st boot? os.conf? static_ip.conf? It is very strange that RPiOS works for Ethernet, but not mEOS 0606 or dev1026.

kner7, 2021-08-23:
I haven't added anything to /boot just installed both discs without messing with them. I haven't done anything besides that what is told in the setup or what you just told me.

I have the ethernet cord plugged straight to my laptop from my pi, that should work just fine right? Or does it have to go through a router?

WiFi works on RPiOS too btw

starbasessd, 2021-08-23:
No, it's not going to work that way, without jumping through a lot of hoops. It needs an IP address from a router, or your laptop need a DHCP server running on that port on the laptop, or both the laptop and the Pi need static IP addresses in the same network range.  If needed, we can go over the wpa_supplicant.conf file at another time. Sounds like the file might be saving as wpa_supplicant.conf.txt or have one of several other common issues.

starbasessd, 2021-08-24:
BTW, was the Raspbian IP address in the 169.254.x.y subnet?

kner7, 2021-08-24:
169.254.122.27 was the raspbian ip adress. I've now set a static ip to the accespoint its connected in the 192.168.1.x subset but still no luck. Im trying to connect it to a LAP-120 access point and connect to it with an Ubiquiti LiteBeam dish

starbasessd, 2021-08-24:
169.254.x.y means it's generating an internal IP address, not one on a network, because it's not seeing a network / valid ethernet connection.
Can you connect it to a router for ethernet?
What are you using to write the image to the SDCard?

kner7, 2021-08-24:
I dont have a router to use :( I'm using a usb micro sd card reader and BalenaEtcher to write to the SDcard

kner7, 2021-08-24:
[image]

this is what the LAP-120 is configured to

kner7, 2021-08-24:
I got it to work!!!, I set the LAP-120 to router mode and turned on DHCP. Thx for steering me in the right direction

closed 2021-08-24 by starbasessd