motioneye-project/motioneyeos #2795: Build instruction missing dosfstools

veldspar, 2021-08-24:
Hey, just a quick note, the build page on the wiki doesnt mention that mkfs.vfat is needed to build the image - whilst minor, would be good to add a simple line like install it with "apt install dosfstools" on a debian-like system. alternatively, build.sh could be fixed to check for mkfs.vfat and install the package automatically...

starbasessd, 2021-08-24:
What was your build environment and which setup instructions did you  use? (OS version, etc)

starbasessd, 2021-08-24:
I just checked sever boxen running Ubuntu 20.04 and earlier, all had mkfs.vfat. Checked Debian Bullseye (11) and no installed mkfs.vfat exfat or fat... Must be a new 'feature'. Checking other versions, but will take a bit. When I confirm the issue, I will modify the docs.

starbasessd, 2021-08-24:
I seem to recall discussion about the DebDevs including a new version of exfat support directly in the kernel, will have to check that too, to see if that causes issues. I know there are issues RPiFoundation still hasn't gotten resolved with Pi-Gen and 32 vs 64 bit build environments and newer versions of Debian.

veldspar, 2021-08-24:
Hey, running debian buster, dosfstools wasn't installed by default, then again I only installed the minimum os and took care of X11 myself. I used the build instructions from the wiki and pulled the source from git repo(because the archived tarball wouldn't build, the kernel and i was too lazy to go into it, i only needed to add wire guard however just changing the kernel on the downloadable image did result in a boot loop bc wifi wouldn't come up and i didn't have time to fix that). I built the image on the raspberry pi platform with a pi zero w as destination.

starbasessd, 2021-08-24:
Interesting, and I wouldn't be surprised a lot of stuff is missing from a minimal Debian on Pi. Out of curiosity, do you remember the build time?

veldspar, 2021-08-24:
Err, i cross compiled from a PC, I should have probably clarified that. Build time was around 60 min on that, but since I do have the resources available I can do a build on a pi as well.

starbasessd, 2021-08-24:
"I built the image on the raspberry pi platform with a pi zero w as destination." confuse(s)(d) me...
I thought you were saying "debian buster, dosfstools wasn't installed by default, then again I only installed the minimum os and took care of X11 myself" on the Pi. 
So you were doing Buster (Lite/server) on X64 cross compiling to PiZeroW?

veldspar, 2021-08-24:
Exactly - i mentioned raspberry pi platform as the build target(since the zero w is included in it). Either way, buster doesn't install dosfstools on default, i just checked by installing a clean instance in VMware with all installer defaults.

starbasessd, 2021-08-24:
Clean instance of Debian 10 Server, or something else?

veldspar, 2021-08-24:
Clean install of debian 10 buster with all installer defaults(sets up x and some ugly wm). Does not come with the dosfstools package by default.

starbasessd, 2021-08-24:
Again, desktop (assuming because you mention x and wm) or server?

veldspar, 2021-08-24:
Desktop.

starbasessd, 2021-08-24:
Modifying instructions, watch for changes.

starbasessd, 2021-08-25:
Changed the documentation at https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source to include dosfstools, and recommend menuconfig (ncurses) for Debian derivatives. Confirmed operation with PiZero image creation. If a user goes to the BuildRoot manual link, they can see QT5 and GTK configuration are still available if using an xserver.

starbasessd, 2021-08-25:
Please close if this resolves your suggestion.