motioneye-project/motioneyeos #2796: Gamma (motioneye or motioneyeos)

Gonioul, 2021-08-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20201026

### Board Model

I am using the following board/model: Pi Zero WH

### Camera

I am using the following type of camera: .

My camera model is: Camera CSI, v1 or FishEye

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi
### Peripherals

I am using the following peripherals that I consider relevant to this issue:
none

### Log Files

I consider the following log files relevant to this issue:
none

### Gamma

Is there a way to apply/change gamma to CSI cameras in network camera mode?
Better if it can be triggered via url via domoticz or automatically for night mode.

starbasessd, 2021-08-24:
If changing between 2 values only, create 2 files one with one value, one with the other.
create 2 scripts to  change the file to the other file
add 2 crontab -e entries to run the 2 scripts 
a\) Set gamma for setting 1
b\) copy file /data/etc/camera-1.conf to /data/etc/camera-1.conf.day
c\) Set gamma for setting 2
d\) copy file /data/etc/camera-1.conf to /data/etc/camera-1.conf.night
e\) nano /data/etc/day.sh
f\) \#\!/bin/sh
    cp /data/etc/camera-1.conf.day /data/etc/camera-1.conf
service motioneye restart
\[ctrl\]x y \[enter\]
g\) nano /data/etc/night.sh
h\) \#\!/bin/sh
    cp /data/etc/camera-1.conf.day /data/etc/camera-1.conf
service motioneye restart
\[ctrl\]x y \[enter\]
i\) crontab -e
j\) add at the end:
0 6 * * * /bin/sh /data/etc/day.sh
0 18 * * * /bin/sh /data/etc/night.sh
\[ctrl\]x y \[enter\]
change the 0 and 6 or 0 and 18 to appropriate times (minute and hour) to run

Gonioul, 2021-08-24:
Well, that's not really the issue, first, I don't know how to change the gamma...
Then, if it's in motion api, domoticz can call the url.

starbasessd, 2021-08-24:
You change the gamma in the WebUI, after changing the  setup to Fast Network Camera, it is stored in the /data/etc/camera-1.conf file.
There isn't an API to call to make the change.

Gonioul, 2021-08-24:
Sorry, I have a lot of video settings like brightness, contrast, but no gamma in the GUI of the fast network camera mode. Nothing in the master motioneye either.
Are you talking about motion's vid_control_params?
I didn't think at first a CSI camera was v4l, but from the cmdline it doesn't handle the gamma setting.
Is there a way to do an optional gamma filter at the camera or at master motioneye?

starbasessd, 2021-08-24:
Is there a single 'Gamma'  setting, no. Gamma isn't simple.
All the settings are available in the Video Device Settings. You would set up the camera to look good for daylight, save as day, then set it for good night vision, and set it to night.
Color, Brightness, Saturation, Contrast.