motioneye-project/motioneyeos #2802: SAMBA Share Fails

Talk2Giuseppe, 2021-08-28:
I am running motionEyeOS version: motionEyeOS 20200606
I am using the following board/model: RPi 3+
I am using the following type of camera: MMAL Camera
My camera model is: Unknown
My motionEyeOS unit is connected to the network via: Wifi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * SanDisk SSD 240 GB

I have multiple instances of MotionEye and various hardware configurations running on the same network.  The cameras which are detecting motion are writing the images to a SanDisk USB storage which is attached to a Raspberry Pi4 running MotionEye in docker on a Debian install.
[image]

My issue is that a configuration running motionEyeOS 20200606 on a Raspberry Pi3 is now suddenly experiencing the inability to locate the shared folder to write the images.  This was a working fine for some time, but I had to recently take the camera down while painters did their thing in my house.  Once the painting was complete, I simply put the camera/pi back into place and plugged it back into the power.
[image]
[image]

Yet, MotionEye running on a Pi Zero with Debian 9.12 has no trouble seeing the shared folder.  This gives me the impression that the issue isn't with the RPi4, but rather with the RPi3 and that version of MotionEyeOS.  But what would cause it to suddenly go sideways?  And most of all, how do I restore the device?  I've done updates, upgrades and reinstalls of Samba.  Nothing seems to be causing it to come back.
[image]

As a result, when MotionEye fails to see the shared folder, the camera will not to display an image.  Rather, you'll get the infamous camera with the slash through it.  This makes it seem like it's a camera feed related issue.
[image]

starbasessd, 2021-08-28:
To clarify, and so I can test, Is this your setup:
Pi4 on static 192.168.1.5/24
Which Base OS? (Debian 9? Why so old?)
Which Docker? (docker-ce or docker.io or other?)
/home/pi/MotionEye shared as MotionEye for Samba (user pi & password \<whatever\>)
Pi3B (or 3B+?) running motionEyeOS 20200606 
Wifi or Ethernet connected? IP address in 192.168.1.x range?
If you ssh to the Pi3, can you ping the Pi4?
While the Pi3 was down, did you run any updates on the Pi4?

Talk2Giuseppe, 2021-08-30:
Thank you for the quick reply!

Pi4 on static 192.168.1.5/24
- Yes... It's running as a sort of server with Home Assistant, MotionEye and PiHole.  HA and MotionEye are in Docker.  My understanding of docker is limited to none.  I had HA and ME running on separate RPi3B+ while PiHole was on a PiZero.  Determined to experiment and learn, I hacked my way around until all 3 were running successfully on a Pi4.  All 3 application co-existed perfectly up until recently.
[image]

Which Base OS? (Debian 9? Why so old?)
- I wouldn't know... And I'm not exactly sure where I pulled that version information from?  Perhaps the zip file?  Anyway, I verified it and its Raspbian v10.
[image]

Which Docker? (docker-ce or docker.io or other?)
- Not sure... See below
[image]

/home/pi/MotionEye shared as MotionEye for Samba (user pi & password <whatever>)
- Yes... And as a reminder, these same mappings were working prior to everything being disconnected for a few weeks.  

Pi3B (or 3B+?) running motionEyeOS 20200606
- 3B+ and Yes
[image]

Wifi or Ethernet connected? IP address in 192.168.1.x range?
- The RPi4 (Server) is Ethernet into the back of the router.  The 3B+ (Dedicated ME) is WiFi.  Yes, the devices are on the same network and in the range of 1-255.

If you ssh to the Pi3, can you ping the Pi4?
- Yes.  I can ping the actual device (no port) and also ping ME (ip:port).  Both reply back.

While the Pi3 was down, did you run any updates on the Pi4?
- Kinda, Sorta, Maybe, Not Sure... I know for a fact I didn't spend any time on the Pi4 while the house was being renovated.  But when I started putting devices back online, issues manifested.  I ran the updates and upgrades on devices which were subject.  The order, process in which things were updated, upgraded, etc is unknown.  So I guess the answer would be yes; sometime between shutting devices down and today, updates were applied.  And as of today, ME is unable to see the shares.  

So as I've been typing up this 4 hour reply, I've been reading posts on similar SAMBA related issues.  Being able to access the shared folders has now become hit or miss from other devices.  Maybe perhaps it's not ME after all.  My suspicion is now related to the RPi4 and the install there.  I was unable to add a new SAMBA user which is weird.  And I am beginning to wonder if perhaps the SanDisk is the issue itself.  In any event, I am going to move the share folders to a newly installed NAS.  That will eliminate the SanDisk and see what happens.  Meanwhile, any other ideas and thoughts would be greatly appreciated.  Thanks!

starbasessd, 2021-08-30:
I couldn't say about the SanDisk. I did toss up a 'quick & dirty' copy of your Pi4 and Pi3B+. I was able to get the Pi4 with latest Raspbian, Samba, Docker-CE (Community Edition) and ME-Docker, and the Pi3B+ with mEOS 20200606 up over the weekend, no issues. 
I was going to suggest checking from a Windows 10 machine, the samba share, but it looks like you have already done so. I would do any backups available of your motionEye, PiHole, and HA settings available, and consider doing a fresh setup and install on the Pi4. It is possible that someone may have killed power to the Pi4 instead of a clean shutdown, and corrupted something that currently shows as issues with the Samba service. (share(s) broken, unable to add user, etc).

Talk2Giuseppe, 2021-08-30:
You may be right about the corrupted file... Boy that sucks!  Yeah, that really sucks!  Rebuilding is pretty time consuming, even with config file backups.  Thanks for your help.  I really appreciate it!

starbasessd, 2021-08-30:
In my environment, I have a tendency to:
Copy the .bash_history file from machines I build to keep a log of what I installed or edited
Base images after setup get imaged to an archive for a restore point
I use Pi-Gen ( a github project from the Raspberry Pi Foundation that they use to create the publicly supported images) to create my base image every other week or so automagically with all my 'normal' mods
I have extensive notes on building RPi images, Debian desktop and server images, Ubuntu desktop and server images, and a few regularly used OSs that I use
I use VirtualBox to create VMs and appliances for most environments
With the exception of 'patches and updates' I seldom take more than an hour to recreate anything.
I use another VM (running apt-cacher-ng) to keep a local copy of all the linux machines latest patches and updates automagically so they're not going out to the internet every time. 
Granted, I do Help Desk (Senior, Enterprise) level support for a living, and have a dedicated VBox server/NAS on my network, but it's doable with an old laptop or seldom used PC (Even a Pi4) too. Depends on how deep into the rabbit hole you want to go...

Talk2Giuseppe, 2021-08-30:
I'm all too familiar with rabbit holes... Hahaha... I have spent almost all of my 56 years of life as a computer/hacker/geek.  Only to find so much frustration in the last several years.  But it's in my DNA and I can't stop tinkering.  Home automation, as fascinating as it is has been a major PIA.  But I am sure I will climb back into that rabbit hole and spend another weekend shelled up at my desk hacking my way around things.  I have so many devices, so much automation that it's a house of cards when it fails.  The wife will immediately destroy any joy I had in building it when she can't turn the LED bulbs back to bright light when the automation fails.  Anyway, thanks for sharing...

In one of your earlier replies, you mentioned a corrupted file may be the cause of the samba issue.  Would uninstalling, removing samba, reboot, reinstall recreate the potentially corrupted files?  Wouldn't that be easier than rebuilding the whole RPi?  I just don't know linux well enough to know if the appropriate/infected/corrupted files were removed.

starbasessd, 2021-08-30:
Simple answer: Possibly. If/when you are ready to rebuild everything else would be the time to check if it would work. It may be more than 1. It may be a bunch. It may not all be in SAMBA services. It's like killing the power when a bunch of files are open in Windows. You're never sure. 
If that is your bent, it can't hurt to try. at least save out your smb.conf before purging SAMBA, then restore it to save some time there...

Talk2Giuseppe, 2021-08-31:
I spent significant time fooling around with SAMBA.  As time went on, nothing was consistent.  It would work for a while, then collapse and disappear.  I would tinker in the smb.conf file a bit and then the share would show up in a file explorer.  I could access one file but not another in the same directory.  Meanwhile, ME would work for a moment, and then not work.  BTW: ME will not show video if the share is failing, AND motion detection is on.  You can have a bad share with motion detection off and the video will display, and that works when settings are done in a certain order.  Anyway, I reinstalled SAMBA several times after a complete purge of SAMBA and SAMBA-COMMON.  In the end, I became convinced SAMBA was causing the problems and not ME.

To prove that theory, I moved the location of the images to a new share on a different device all together.  And within seconds, everything fell in place.  So that, in my mind, pretty much removed ME as the problem.  Question now is trying to figure out why SAMBA is failing on the RPi4.  My suspicion now is that the OS on the Pi4 is probably at fault as it's the only thing that experienced a change since resurrecting the camera network... Unless of course, SAMBA was also updated/upgraded at the same time.

starbasessd, 2021-09-10:
Anything further?

Talk2Giuseppe, 2021-09-10:
I haven't spent any additional time on trying to fix SAMBA on the RPi 4.  Not sure why it stopped working.  I now have the images being saved directly to a stand alone NAS.  I'll eventually need to give it some attention cause it's messing up my ability to edit Home Assistant configuration files.  Some searching has confirmed that this is an issue with SAMBA. 

Basically, none of this issue is the result of MotionEye itself so I figured I'd end this thread.

starbasessd, 2021-09-10:
Please go ahead and close then, thanks.

closed 2021-09-10 by Talk2Giuseppe