motioneye-project/motioneyeos #2803: NoIR Unable to Open Video Device

jammininmypjs, 2021-09-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20200606
### Board Model

I am using the following board/model: RPI Zero W

### Camera

I am using the following type of camera: NoIR

My camera model is: v2.1

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A (unless Samba shares count? In which case it's a WD My Book)

### Log Files

I consider the following log files relevant to this issue:
[motion.log](https://github.com/ccrisan/motioneyeos/files/7112350/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/7112351/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/7112352/boot.log)

I saw a similar issue open in motioneye, but not here. I am able to add the camera as a V4L2 device, but not MMAL. I've tried two different cables and I used to use this camera so it should work. Could the camera need enabled on the pi or something? Or should I just get a new camera

starbasessd, 2021-09-05:
In motionEyeOS it's automatically enabled, and should self install. 
The Black or blue side of the cable goes away from the board.
If you can ssh into it, run the command:
vcgencmd get_camera
and report the result here.

starbasessd, 2021-09-05:
BTW, any CSI camera is going to be a MMAL camera.

jammininmypjs, 2021-09-05:
supported=1 detected=0
Ah so looks like perhaps a new cable or camera since it's just not being detected

starbasessd, 2021-09-05:
Or cables aren't connected correctly.

jammininmypjs, 2021-09-05:
Nah I've tried disconnecting and reconnecting a couple times and made sure theyre facing the right way. I'll test with a cable I know works and see what happens. Thanks!