motioneye-project/motioneyeos #2805: Camera URL too long can it be lengthened/edited?

qxotic, 2021-09-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

motionEye Version | 0.42.1
-- | --
Motion Version | 4.1.1
OS Version | Linux 4.14.241+

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
running on DietPi v7.5.2 : 07:41 - Wed 09/08/2021 on an Odroid HC1
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
Network Camera Ansice 

My camera model is: (insert your camera model, e.g. Logitech C290).

Ansice POE IP Camera IR Infrared 4.0MP H.265 PoE 4MP Black Metal Dome Network Camera IP System Weatherproof IP66 (3.6mm Lens)  from Amazon
### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
Ethernet
### Peripherals

I am using the following peripherals that I consider relevant to this issue:
None
 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files   relevant from motion.log:
[5:ml5:Camera5] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera5): Invalid data found when processing input
[5:ml5:Camera5] [ERR] [VID] vid_start: Netcam RTSP failed to open
[5:ml5:Camera5] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[5:ml5:Camera5] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[5:ml5:Camera5] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[5:ml5:Camera5] [ERR] [NET] netcam_rtsp_open_context: Normal resolution: Unable to open camera(Camera5): Invalid data found when processing input

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
As you can tell, this is camera 5... Other cameras are up and operating fine.. all different models from each other.  I am able to get the Ansice camera to stream in VLC using:  

rtsp://admin:@10.0.0.37:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream

I noticed when trying to add in motioneye even when I take out the user and password information there is not enough room for the URL.  rtsp://10.0.0.37:554/channel=1_stream=1.sdp?real_stream   is all the longer the field is.  I am not able to include the =0.sdp?real_stream.  Can this be changed, or edited somewhere else to allow for the full url?  I believe this is why I am unable to get this camera to work.  This camera only sends two rtsp urls, and they both follow the same format and are the same length.  The only difference is stream 0 and 1.

starbasessd, 2021-09-08:
Unable to locate manufacturer "Ansice" on the iSpy database:
https://www.ispyconnect.com/cameras/A
Have you trued editing the appropriate line in:
/etc/motioneye/camera-5.conf?

qxotic, 2021-09-08:
Thank you for that information, it pointed me to the config file.  Actually, the entire url was in the config, it just does not display the whole line when adding the camera (I had cut and pasted it to add it)  (It does display it in the camera menu after adding if you scroll over).   The line was correct... the camera was not working.. but... I had an idea and tried replacing the url with the one that worked in VLC.. the one that included the user and password.. and it worked!   I edited the config file and changed the url to :  
rtsp://admin:@10.0.0.37:554/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp?real_stream

I also edited the width in there to the odd 704 width this camera uses.  
I rebooted the Odroid and when I logged back in the camera was displaying.

Thank you!

starbasessd, 2021-09-08:
Glad to point you in the right direction.

closed 2021-09-08 by starbasessd